embassy-rs/embassy #3784: Run button_exti example on STM32H5 but a panic occurs

joshuajylin, 2025-01-20:
I guess there's something wrong when MCU wakes up from the low-power mode.
I added button_exti into the stop example and everything worked fine.

`cargo run --bin button_exti`
`    Finished `dev` profile [optimized + debuginfo] target(s) in 0.39s
     Running `probe-rs run --chip STM32H563ZITx target/thumbv8m.main-none-eabihf/debug/button_exti`
      Erasing ✔ [00:00:00] [####################################################################################################################################################################################] 32.00 KiB/32.00 KiB @ 411.40 KiB/s (eta 0s )
  Programming ✔ [00:00:00] [#####################################################################################################################################################################################] 32.00 KiB/32.00 KiB @ 65.86 KiB/s (eta 0s )    Finished in 0.578s
DEBUG flash: latency=1 wrhighfreq=0
└─ embassy_stm32::rcc::_version::flash_setup @ /Users/joshua/Workspace/github/embassy/embassy-stm32/src/rcc/h.rs:1000
TRACE BDCR ok: 0c008200
└─ embassy_stm32::rcc::bd::{impl#3}::init @ /Users/joshua/Workspace/github/embassy/embassy-stm32/src/rcc/bd.rs:216 
DEBUG rcc: Clocks { audioclk: MaybeHertz(0), csi: MaybeHertz(0), csi_div_122: MaybeHertz(0), hclk1: MaybeHertz(64000000), hclk2: MaybeHertz(64000000), hclk4: MaybeHertz(64000000), hse: MaybeHertz(0), hsi: MaybeHertz(64000000), hsi48: MaybeHertz(48000000), lse: MaybeHertz(0), lsi: MaybeHertz(0), pclk1: MaybeHertz(64000000), pclk1_tim: MaybeHertz(64000000), pclk2: MaybeHertz(64000000), pclk2_tim: MaybeHertz(64000000), pclk3: MaybeHertz(64000000), pll1_q: MaybeHertz(0), pll2_p: MaybeHertz(0), pll2_q: MaybeHertz(0), pll2_r: MaybeHertz(0), pll3_p: MaybeHertz(0), pll3_q: MaybeHertz(0), pll3_r: MaybeHertz(0), rtc: MaybeHertz(32000), sys: MaybeHertz(64000000) }
└─ embassy_stm32::rcc::set_freqs @ /Users/joshua/Workspace/github/embassy/embassy-stm32/src/rcc/mod.rs:71  
INFO  Hello World!
└─ button_exti::____embassy_main_task::{async_fn#0} @ src/bin/button_exti.rs:13  
INFO  Press the USER button...
└─ button_exti::____embassy_main_task::{async_fn#0} @ src/bin/button_exti.rs:17  
ERROR panicked at /Users/joshua/Workspace/github/embassy/embassy-stm32/src/low_power.rs:97:23:
called `Option::unwrap()` on a `None` value
└─ panic_probe::print_defmt::print @ /Users/joshua/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.2/src/lib.rs:104`

joshuajylin, 2025-01-20:
I found that low_power::on_wakeup_irq() will be called if the low-power feature is enabled.
So I removed the low-power feature from embassy-stm32 in Cargo.toml and the button_exti example worked.

yodaldevoid, 2025-04-09:
I have also seen this on an STM32H533RE in (nearing) production code. All that was required was having the `low-power` feature enabled, wait until all tasks are pending, wait some significant amount of time, and then do something that would cause the system to wake up.