embassy-rs/embassy #3789: The functions set_using_xosc(), set_using_lposc(), set_using_gpio_1khz(), set_using_gpio_1hz() in \src\rp235x\powman\regs.rs do not make sense

mschnell1, 2025-01-21:
as the appropriate bits in the TIMER register are read only.

mschnell1, 2025-01-27:
Moreover IMHO,  the function  `set_alarm(&mut self, val: bool) ` should be called `reset_alarm()` and have no argument, as the appropriate bit in the `TIMER `register is "Clear on write 1".

mschnell1, 2025-01-27:
BTW.: it would be nice if the "Register" functions would support the "Constructor Pattern". 
such as:
```
TIMER.timer()
  .read()
  .set_nonsec_write()
  .clear()
  .set_alarm_enab(false)
  .reset_alarm()
  .set_key()
  .write_value()
```

With set_key setting the upper 16 bits to  `0x5afe`

Dirbaio, 2025-01-27:
the PAC is autogenerated by [chiptool](https://github.com/embassy-rs/chiptool/). It's not possible to do ad-hoc changes to individual registers like that, they all have to work the same way.

It's intentional that all fields are readable/writable. The "register value" structs are a copy of the register in memory. Think of it like a "u32 but with a nice wrapper to get/set all the fields".

Chiptool intentionally doesn't support "chaining" method calls like that because it results in less readable code.

mschnell1, 2025-01-27:
I don't agree, but I do see your meaning.
Thanks for answering !

mschnell1, 2025-01-28:
I found that the buider Pattern does not as much sense, as I supposed, because  the write function's target which needs to be re-evaluated.

closed 2025-01-28 by mschnell1