embassy-rs/embassy #3798: embassy_stm32::init() hangs using STM32G07

jduranf, 2025-01-22:
I'm using a NUCLEO-G071RB and embassy-stm32 version 0.2.0. The code compiles fine, but hangs inside of `embassy_stm32::init(Default::default())`, in hsi48.rs line 44 `while r.read().hsi48rdy() == false {}`. If I switch back to version 0.1.0, everything works fine.

My main.rs:

```rust
#![no_std]
#![no_main]

use defmt::*;
use panic_probe as _;
use defmt_rtt as _;
use cortex_m_rt::entry;

#[entry]
fn main() -> ! {
    info!("Let's go!");
    embassy_stm32::init(Default::default());
    info!("Never reached :(");

    loop {};
}
```

Cargo.toml:

```toml
[package]
name = "embassy"
version = "0.1.0"
edition = "2021"

[dependencies]
embassy-stm32 = { version = "0.2.0", features = ["defmt", "time-driver-any", "stm32g0b1re", "memory-x", "unstable-pac", "exti"] }
embassy-executor = { version = "0.7.0", features = ["defmt", "arch-cortex-m", "executor-thread"] }
embassy-time = { version = "0.4.0", features = ["defmt", "tick-hz-32_768", "defmt-timestamp-uptime"] }

cortex-m = { version = "0.7.7", features = ["critical-section-single-core"] }
cortex-m-rt = "0.7.5"
defmt = "0.3.10"
defmt-rtt = "0.4.1"
panic-probe = { version = "0.3.2", features = ["print-defmt"] }
```

And [all the project](https://github.com/user-attachments/files/18510837/embassy.zip)

itswenb, 2025-01-24:
Since you are using `embassy-executor` with feature `executor-thread`, then the entrance should be annotated with `#[embassy_executor::main]` instead.

Dirbaio, 2025-01-24:
> embassy-stm32 = { version = "0.2.0", features = ["defmt", "time-driver-any", "stm32g0b1re", "memory-x", "unstable-pac", "exti"] }

change stm32g0b1re to stm32g071rb

jduranf, 2025-01-24:
Thank you @Dirbaio! This fixed the issue. I feel a little bit stupid...

closed 2025-01-24 by jduranf