embassy-rs/embassy #3805: ADC interrupt for STM32F103 doesn't work property

Tnze, 2025-01-25:
Current ADC implement only works on the example. Adding totally unrelated parts to the program breaks the ADC function.

- The default sample time `SampleTime::CYCLES1_5` works. Set to other cause `Adc::read().await` hang-up forever.
- Add line `defmt::info!("convert started");` to `embassy-stm32/src/adc/f1.rs` file `convert()` function above line `T::state().waker.register(....` makes it work...
- My project uses USB CDC. When start the USB task, ADC doesn't work, when remove USB stuffs the ADC works again.

, which is quite confusing. And it takes my hours to debug.

It turns out that the ADC End of Convert interrupt doesn't triggered at all. So the `poll_fn` wouldn't wake up once it return `Poll::Pending`. It only works if the convert was completed before the `poll_fn` is polled.

https://github.com/embassy-rs/embassy/blob/51d87c6603631fda6fb59ca1a65a99c08138b081/embassy-stm32/src/adc/f1.rs#L124C1-L132C11

Tnze, 2025-01-25:
Alos, the example `stm32f1/src/bin/adc.rs` only works on `Debug`. Release doesn't work.

I think it's because of `Debug` runs a lot slowly, so that the ADC can finish its work before the HAL first query the `eof` flag.

Tnze, 2025-01-25:
Regards @xoviat ,
I would be grateful if you could help. I'm unable to find the reason why the EOC interrupt didn't be triggered.

Tnze, 2025-01-25:
Add this line fix this problem:
```rust
unsafe { NVIC::unmask(pac::interrupt::ADC1_2) };
```

Dirbaio, 2025-04-06:
fixed in #4056

closed 2025-04-06 by Dirbaio