embassy-rs/embassy #3815: Unknown USB Device (Invalid Configuration Descriptor)

dimpolo, 2025-01-27:
I'm trying to get stm32u595ri USB communication working.
After applying these [changes](https://github.com/embassy-rs/embassy/commit/ab494b432674a708aaf17e9e8f0c64af34100515) I'm running the following:
```
cd examples/stm32u5
cargo r -r --bin usb_hs_serial
```
On Windows 10 I'm seeing:

[image]

And this inside Device Manager:
[image]

I'm assuming it's not a hardware issue. This is because tying BOOT0 to VDD successfully allows me to interface with the chip using the USB DFU bootloader.

<details>
<summary>Here's the logs:</summary>

```
PS C:\Users\user_name\embassy\examples\stm32u5> cargo r -r --bin usb_hs_serial
    Finished `release` profile [optimized + debuginfo] target(s) in 0.23s
     Running `probe-rs run --chip STM32U595RITx target\thumbv8m.main-none-eabihf\release\usb_hs_serial`
      Erasing ✔ [00:00:00] [##########################################################################################################################################################] 56.00 KiB/56.00 KiB @ 394.37 KiB/s (eta 0s )
  Programming ✔ [00:00:01] [###########################################################################################################################################################] 52.00 KiB/52.00 KiB @ 28.91 KiB/s (eta 0s )    Finished in 1.967s
 WARN defmt_decoder::log::format: logger format contains timestamp but no timestamp implementation was provided; consider removing the timestamp (`{t}` or `{T}`) from the logger format or provide a `defmt::timestamp!` implementation    
0.000000 INFO  Hello World!
└─ usb_hs_serial::____embassy_main_task::{async_fn#0} @ src\bin\usb_hs_serial.rs:21  
0.000000 TRACE BDCR ok: 0c00a200
└─ embassy_stm32::rcc::bd::{impl#3}::init @ C:\Users\user_name\embassy\embassy-stm32\src\rcc\bd.rs:216 
0.000000 DEBUG rcc: Clocks { audioclk: MaybeHertz(0), hclk1: MaybeHertz(160000000), hclk2: MaybeHertz(160000000), hclk3: MaybeHertz(160000000), hse: MaybeHertz(16000000), hse_div_2: MaybeHertz(8000000), hsi: MaybeHertz(0), hsi48: MaybeHertz(48000000), hsi48_div_2: MaybeHertz(0), lse: MaybeHertz(0), lsi: MaybeHertz(32000), msik: MaybeHertz(4000000), pclk1: MaybeHertz(160000000), pclk1_tim: MaybeHertz(160000000), pclk2: MaybeHertz(160000000), pclk2_tim: MaybeHertz(160000000), pclk3: MaybeHertz(160000000), pll1_p: MaybeHertz(32000000), pll1_p_div_2: MaybeHertz(16000000), pll1_q: MaybeHertz(48000000), pll2_p: MaybeHertz(0), pll2_q: MaybeHertz(0), pll2_r: MaybeHertz(0), pll3_p: MaybeHertz(0), pll3_q: MaybeHertz(0), rtc: MaybeHertz(32000), sys: MaybeHertz(160000000) }
└─ embassy_stm32::rcc::set_freqs @ C:\Users\user_name\embassy\embassy-stm32\src\rcc\mod.rs:71  
0.000000 TRACE allocating type=Interrupt mps=8 interval_ms=255, dir=In
└─ embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:351 
0.000030 TRACE   index=1
└─ embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:407
0.000061 TRACE allocating type=Bulk mps=64 interval_ms=0, dir=Out
└─ embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:351
0.000091 TRACE   index=1
└─ embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:407
0.000122 TRACE allocating type=Bulk mps=64 interval_ms=0, dir=In
└─ embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:351
0.000152 TRACE   index=2
└─ embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:407
0.000183 INFO  USB: config_descriptor used: 70
└─ embassy_usb::builder::{impl#1}::build @ C:\Users\user_name\embassy\embassy-usb\src\builder.rs:221
0.000183 INFO  USB: bos_descriptor used: 12
└─ embassy_usb::builder::{impl#1}::build @ C:\Users\user_name\embassy\embassy-usb\src\builder.rs:222
0.000213 INFO  USB: msos_descriptor used: 0
└─ embassy_usb::builder::{impl#1}::build @ C:\Users\user_name\embassy\embassy-usb\src\builder.rs:223
0.000213 INFO  USB: control_buf size: 64
└─ embassy_usb::builder::{impl#1}::build @ C:\Users\user_name\embassy\embassy-usb\src\builder.rs:224
0.000244 TRACE allocating type=Control mps=64 interval_ms=0, dir=Out
└─ embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:351
0.000274 TRACE   index=0
└─ embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:407
0.000305 TRACE allocating type=Control mps=64 interval_ms=0, dir=In
└─ embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:351
0.000335 TRACE   index=0
└─ embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:407
0.000335 TRACE start
└─ embassy_usb_synopsys_otg::{impl#8}::start @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:466
0.000366 TRACE Core id 00005000
└─ embassy_stm32::usb::_version::{impl#3}::init @ C:\Users\user_name\embassy\embassy-stm32\src\usb\otg.rs:332
0.000396 TRACE usb: power detected
└─ embassy_usb::{impl#2}::handle_bus_event::{async_fn#0} @ C:\Users\user_name\embassy\embassy-usb\src\lib.rs:480
0.000396 TRACE enable
└─ embassy_usb_synopsys_otg::{impl#10}::enable::{async_fn#0} @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:964
0.000427 TRACE irq
└─ embassy_usb_synopsys_otg::on_interrupt @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:29
0.000427 TRACE vbus detected
[... 341 lines cut ...]
2.111328 TRACE === status 01850000
└─ embassy_usb_synopsys_otg::on_interrupt @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:45
2.111358 TRACE OUT_DATA_RX ep=0 len=0
└─ embassy_usb_synopsys_otg::on_interrupt @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:71
2.111389 TRACE === status 01870000
└─ embassy_usb_synopsys_otg::on_interrupt @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:45
2.111389 TRACE OUT_DATA_DONE ep=0
└─ embassy_usb_synopsys_otg::on_interrupt @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:98
2.111419 TRACE read ep=EndpointAddress(0) done len=0
└─ embassy_usb_synopsys_otg::{impl#15}::read::{async_fn#0}::{closure#0} @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:1069
2.111450 TRACE control: complete
└─ embassy_usb_synopsys_otg::{impl#17}::data_in::{async_fn#0} @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:1285
2.111450 TRACE SETUP waiting
└─ embassy_usb_synopsys_otg::{impl#17}::setup::{async_fn#0}::{closure#0} @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:1263
2.111480 TRACE SETUP waiting
└─ embassy_usb_synopsys_otg::{impl#17}::setup::{async_fn#0}::{closure#0} @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:1263
2.117645 TRACE irq
└─ embassy_usb_synopsys_otg::on_interrupt @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:29
2.117675 TRACE suspend
└─ embassy_usb_synopsys_otg::{impl#10}::poll::{async_fn#0}::{closure#0} @ C:\Users\user_name\embassy\embassy-usb-synopsys-otg\src\lib.rs:834
2.117675 TRACE usb: suspend
└─ embassy_usb::{impl#2}::handle_bus_event::{async_fn#0} @ C:\Users\user_name\embassy\embassy-usb\src\lib.rs:473

```
</details>

dimpolo, 2025-02-10:
@mubes @MDr164 any ideas where to look?

guptaarnav, 2025-02-10:
@dimpolo I don't have a u5 mcu to test this out on but could you try replacing PLL1_P to PLL_Q on line 39 of examples/stm32u5/usb_hs_serial.rs?  Could be a timing issue with the PLL setup.  this is a shot in the dark, may not fix it.

// Use PLL1_Q (48MHz) for the USB clock.
config.rcc.mux.otghssel = mux::Otghssel::PLL1_Q;

MDr164, 2025-02-10:
I only tried this out on the Nucleo so far as the project I'm building with that chip is not yet done on the hw design side and also only got a Linux machine here to test. It does look like the USB stack is coming up just fine but the descriptor looks off. Can you check from Windows what it actually sees and if it matches your configured USB descriptor?

dimpolo, 2025-02-11:
Thanks a lot for the responses!

@guptaarnav PLL1_Q is not a setting I can select. The available values are:
```rust
    pub enum Otghssel {
        #[doc = "HSE selected"]
        HSE = 0x0,
        #[doc = "PLL1 “P” (pll1_q_ck) selected,"]
        PLL1_P = 0x01,
        #[doc = "HSE/2 selected"]
        HSE_DIV_2 = 0x02,
        #[doc = "PLL1 “P” divided by 2 (pll1_p_ck/2) selected"]
        PLL1_P_DIV_2 = 0x03,
    }
```

@MDr164 I'll try to find this information and get back at you

dimpolo, 2025-02-12:
@MDr164 
I tried to extract some useful information from windows but I couldn't find anything. I tried using wireshark to log usb coms but nothing came up.

Next I tried using a Linux machine, but I'm not familiar with it at all.

<details>
<summary>Using usbmon I got these logs</summary>

```
ffff9f23d1a21600 2730281021 C Ii:2:001:1 0:2048 2 = 0800
ffff9f23d1a21600 2730281039 S Ii:2:001:1 -115:2048 4 <
ffff9f247e095540 2730281053 S Ci:2:001:0 s a3 00 0000 0003 0004 4 <
ffff9f247e095540 2730281063 C Ci:2:001:0 0 4 = 01010100
ffff9f247e095540 2730281067 S Co:2:001:0 s 23 01 0010 0003 0000 0
ffff9f247e095540 2730281072 C Co:2:001:0 0 0
ffff9f247e095540 2730281075 S Ci:2:001:0 s a3 00 0000 0003 0004 4 <
ffff9f247e095540 2730281079 C Ci:2:001:0 0 4 = 01010000
ffff9f247e095540 2730316642 S Ci:2:001:0 s a3 00 0000 0003 0004 4 <
ffff9f247e095540 2730316673 C Ci:2:001:0 0 4 = 01010000
ffff9f247e095540 2730352666 S Ci:2:001:0 s a3 00 0000 0003 0004 4 <
ffff9f247e095540 2730352690 C Ci:2:001:0 0 4 = 01010000
ffff9f247e095540 2730388711 S Ci:2:001:0 s a3 00 0000 0003 0004 4 <
ffff9f247e095540 2730388743 C Ci:2:001:0 0 4 = 01010000
ffff9f247e095540 2730424622 S Ci:2:001:0 s a3 00 0000 0003 0004 4 <
ffff9f247e095540 2730424648 C Ci:2:001:0 0 4 = 01010000
ffff9f247e095540 2730424762 S Co:2:001:0 s 23 03 0004 0003 0000 0
ffff9f247e095540 2730424776 C Co:2:001:0 0 0
ffff9f247e095540 2730492706 S Ci:2:001:0 s a3 00 0000 0003 0004 4 <
ffff9f247e095540 2730492739 C Ci:2:001:0 0 4 = 03051000
ffff9f247e095540 2730492753 S Co:2:001:0 s 23 01 0014 0003 0000 0
ffff9f247e095540 2730492767 C Co:2:001:0 0 0
ffff9f247e095540 2730552708 S Ci:2:000:0 s 80 06 0100 0000 0040 64 <
ffff9f247e095540 2730553220 C Ci:2:000:0 0 18 = 12011002 ef020140 c016dd27 10000102 0301
ffff9f247e095540 2730553245 S Co:2:001:0 s 23 03 0004 0003 0000 0
ffff9f247e095540 2730553263 C Co:2:001:0 0 0
ffff9f247e095540 2730620620 S Ci:2:001:0 s a3 00 0000 0003 0004 4 <
ffff9f247e095540 2730620628 C Ci:2:001:0 0 4 = 03051000
ffff9f247e095540 2730620632 S Co:2:001:0 s 23 01 0014 0003 0000 0
ffff9f247e095540 2730620637 C Co:2:001:0 0 0
ffff9f247e095540 2730700680 S Ci:2:008:0 s 80 06 0100 0000 0012 18 <
ffff9f247e095540 2730701281 C Ci:2:008:0 0 18 = 12011002 ef020140 c016dd27 10000102 0301
ffff9f247e095540 2730701324 S Ci:2:008:0 s 80 06 0f00 0000 0005 5 <
ffff9f247e095540 2730701926 C Ci:2:008:0 0 5 = 050f0c00 01
ffff9f247e095540 2730701969 S Ci:2:008:0 s 80 06 0f00 0000 000c 12 <
ffff9f247e095540 2730702591 C Ci:2:008:0 0 12 = 050f0c00 01071002 00000000
ffff9f247e095540 2730702641 S Ci:2:008:0 s 80 06 0200 0000 0009 9 <
ffff9f247e095540 2730703197 C Ci:2:008:0 0 9 = 09024600 02010080 32
ffff9f247e095540 2730703242 S Ci:2:008:0 s 80 06 0200 0000 0046 70 <
ffff9f247e095540 2730704315 C Ci:2:008:0 0 70 = 09024600 02010080 32080b00 02020200 00090400 00010202 00000524 00100104
ffff9f247e0950c0 2730704404 S Ci:2:008:0 s 80 06 0300 0000 00ff 255 <
ffff9f247e0950c0 2730704837 C Ci:2:008:0 0 4 = 04030904
ffff9f247e0950c0 2730704877 S Ci:2:008:0 s 80 06 0302 0409 00ff 255 <
ffff9f247e0950c0 2730705561 C Ci:2:008:0 0 38 = 26035500 53004200 2d007300 65007200 69006100 6c002000 65007800 61006d00
ffff9f247e0950c0 2730705586 S Ci:2:008:0 s 80 06 0301 0409 00ff 255 <
ffff9f247e0950c0 2730706203 C Ci:2:008:0 0 16 = 10034500 6d006200 61007300 73007900
ffff9f247e0950c0 2730706228 S Ci:2:008:0 s 80 06 0303 0409 00ff 255 <
ffff9f247e0950c0 2730706917 C Ci:2:008:0 0 18 = 12033100 32003300 34003500 36003700 3800
ffff9f247e0950c0 2730708903 S Co:2:008:0 s 00 09 0001 0000 0000 0
ffff9f247e0950c0 2730709384 C Co:2:008:0 0 0
ffff9f23cfec4600 2730710197 S Co:2:008:0 s 21 20 0000 0000 0007 7 = 80250000 000008
ffff9f23cfec4600 2730710863 C Co:2:008:0 0 7 >

```
</details>

Do you have a recommendation for how to proceed?

dimpolo, 2025-02-12:
Wow, so under Linux all is working fine.
Do you happen to know what I could look into to get Windows working?

MDr164, 2025-02-12:
Most of the USB examples leave the msos descriptor empty. You probably need that one for Windows to correctly detect your device. If you look at the example in the u5 folder try adding an msos descriptor the same way the bos descriptor is added.

dimpolo, 2025-02-12:
I added the config lines. This resulted in `ERROR panicked at 'if composite_with_iads is set, you must set device_class = 0xEF, device_sub_class = 0x02, device_protocol = 0x01'` so I added `config.composite_with_iads = false;` Unfortunaltly this did not change anything.

Sorry I don't quite follow about the msos. The only examples I found that make use of msos seem to be WINUSB related. Should I copy those?

reed38, 2025-02-19:
I'm having a similar issue with stm32u5a5zj chip on the NUCLEO U5A5ZJ-Q dev board.

Here is are the logs:
```log
[INFO ][0.000000][usb_hs_serial::____embassy_main_task::{async_fn#0}][usb_hs_serial.rs][21  ]Hello World!
[TRACE][0.000000][embassy_stm32::rcc::bd::{impl#3}::init][bd.rs][216 ]BDCR ok: 0c00a200
[DEBUG][0.000000][embassy_stm32::rcc::set_freqs][mod.rs][71  ]rcc: Clocks { audioclk: MaybeHertz(0), hclk1: MaybeHertz(160000000), hclk2: MaybeHertz(160000000), hclk3: MaybeHertz(160000000), hse: MaybeHertz(16000000), hse_div_2: MaybeHertz(8000000), hsi: MaybeHertz(0), hsi48: MaybeHertz(48000000), hsi48_div_2: MaybeHertz(0), lse: MaybeHertz(0), lsi: MaybeHertz(32000), msik: MaybeHertz(4000000), pclk1: MaybeHertz(160000000), pclk1_tim: MaybeHertz(160000000), pclk2: MaybeHertz(160000000), pclk2_tim: MaybeHertz(160000000), pclk3: MaybeHertz(160000000), pll1_p: MaybeHertz(32000000), pll1_p_div_2: MaybeHertz(16000000), pll1_q: MaybeHertz(48000000), pll2_p: MaybeHertz(0), pll2_q: MaybeHertz(0), pll2_r: MaybeHertz(0), pll3_p: MaybeHertz(0), pll3_q: MaybeHertz(0), rtc: MaybeHertz(32000), shsi: MaybeHertz(0), shsi_div_2: MaybeHertz(0), sys: MaybeHertz(160000000) }
[TRACE][0.000122][embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint][lib.rs][351 ]allocating type=Interrupt mps=8 interval_ms=255, dir=In
[TRACE][0.000366][embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint][lib.rs][407 ]  index=1
[TRACE][0.000549][embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint][lib.rs][351 ]allocating type=Bulk mps=64 interval_ms=0, dir=Out
[TRACE][0.000793][embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint][lib.rs][407 ]  index=1
[TRACE][0.000915][embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint][lib.rs][351 ]allocating type=Bulk mps=64 interval_ms=0, dir=In
[TRACE][0.001190][embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint][lib.rs][407 ]  index=2
[INFO ][0.001342][embassy_usb::builder::{impl#1}::build][builder.rs][221 ]USB: config_descriptor used: 70
[INFO ][0.001464][embassy_usb::builder::{impl#1}::build][builder.rs][222 ]USB: bos_descriptor used: 12
[INFO ][0.001586][embassy_usb::builder::{impl#1}::build][builder.rs][223 ]USB: msos_descriptor used: 0
[INFO ][0.001708][embassy_usb::builder::{impl#1}::build][builder.rs][224 ]USB: control_buf size: 64
[TRACE][0.001831][embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint][lib.rs][351 ]allocating type=Control mps=64 interval_ms=0, dir=Out
[TRACE][0.002075][embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint][lib.rs][407 ]  index=0
[TRACE][0.002166][embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint][lib.rs][351 ]allocating type=Control mps=64 interval_ms=0, dir=In
[TRACE][0.002410][embassy_usb_synopsys_otg::{impl#7}::alloc_endpoint][lib.rs][407 ]  index=0
[TRACE][0.002532][embassy_usb_synopsys_otg::{impl#8}::start][lib.rs][466 ]start
[TRACE][0.002746][embassy_stm32::usb::_version::{impl#3}::init][otg.rs][332 ]Core id 00005000
[TRACE][0.002868][embassy_usb::{impl#2}::handle_bus_event::{async_fn#0}][lib.rs][480 ]usb: power detected
[TRACE][0.002960][embassy_usb_synopsys_otg::{impl#10}::enable::{async_fn#0}][lib.rs][964 ]enable
[TRACE][0.003082][embassy_usb_synopsys_otg::{impl#17}::setup::{async_fn#0}::{closure#0}][lib.rs][1263]SETUP waiting
[TRACE][0.009002][embassy_usb_synopsys_otg::on_interrupt][lib.rs][29  ]irq
[TRACE][0.009124][embassy_usb_synopsys_otg::{impl#10}::poll::{async_fn#0}::{closure#0}][lib.rs][834 ]suspend
[TRACE][0.009216][embassy_usb::{impl#2}::handle_bus_event::{async_fn#0}][lib.rs][473 ]usb: suspend
[TRACE][0.103729][embassy_usb_synopsys_otg::on_interrupt][lib.rs][29  ]irq
[TRACE][0.103881][embassy_usb_synopsys_otg::{impl#10}::poll::{async_fn#0}::{closure#0}][lib.rs][803 ]reset
[TRACE][0.103973][embassy_usb_synopsys_otg::{impl#9}::init_fifo][lib.rs][638 ]init_fifo
[TRACE][0.104095][embassy_usb_synopsys_otg::{impl#9}::init_fifo::{closure#0}][lib.rs][647 ]configuring rx fifo size=62
[TRACE][0.104248][embassy_usb_synopsys_otg::{impl#9}::init_fifo::{closure#0}][lib.rs][655 ]configuring tx fifo ep=0, offset=62, size=16
[TRACE][0.104461][embassy_usb_synopsys_otg::{impl#9}::init_fifo::{closure#0}][lib.rs][655 ]configuring tx fifo ep=1, offset=78, size=16
[TRACE][0.104675][embassy_usb_synopsys_otg::{impl#9}::init_fifo::{closure#0}][lib.rs][655 ]configuring tx fifo ep=2, offset=94, size=16
[TRACE][0.104919][embassy_usb_synopsys_otg::{impl#9}::configure_endpoints][lib.rs][695 ]configure_endpoints
[TRACE][0.158508][embassy_usb_synopsys_otg::on_interrupt][lib.rs][29  ]irq
[TRACE][0.158660][embassy_usb_synopsys_otg::{impl#10}::poll::{async_fn#0}::{closure#0}][lib.rs][820 ]enumdne
[TRACE][0.158752][embassy_usb_synopsys_otg::{impl#10}::poll::{async_fn#0}::{closure#0}][lib.rs][824 ]  speed=0 trdt=9
[TRACE][0.158935][embassy_usb::{impl#2}::handle_bus_event::{async_fn#0}][lib.rs][447 ]usb: reset
[TRACE][0.159088][embassy_usb_synopsys_otg::{impl#17}::setup::{async_fn#0}::{closure#0}][lib.rs][1263]SETUP waiting
[TRACE][0.191436][embassy_usb_synopsys_otg::on_interrupt][lib.rs][29  ]irq
[TRACE][0.191558][embassy_usb_synopsys_otg::on_interrupt][lib.rs][45  ]=== status 00ac0080
[TRACE][0.191711][embassy_usb_synopsys_otg::on_interrupt][lib.rs][53  ]SETUP_DATA_RX
[TRACE][0.191802][embassy_usb_synopsys_otg::on_interrupt][lib.rs][45  ]=== status 00a80000
[TRACE][0.191955][embassy_usb_synopsys_otg::on_interrupt][lib.rs][101 ]SETUP_DATA_DONE ep=0
[TRACE][0.192108][embassy_usb_synopsys_otg::on_interrupt][lib.rs][29  ]irq
[... cut ...]
[TRACE][2.318084][embassy_usb_synopsys_otg::{impl#17}::setup::{async_fn#0}::{closure#0}][lib.rs][1263]SETUP waiting
[TRACE][2.318237][embassy_usb_synopsys_otg::{impl#17}::setup::{async_fn#0}::{closure#0}][lib.rs][1263]SETUP waiting
[TRACE][2.322662][embassy_usb_synopsys_otg::on_interrupt][lib.rs][29  ]irq
[TRACE][2.322814][embassy_usb_synopsys_otg::{impl#10}::poll::{async_fn#0}::{closure#0}][lib.rs][834 ]suspend
[TRACE][2.322906][embassy_usb::{impl#2}::handle_bus_event::{async_fn#0}][lib.rs][473 ]usb: suspend

```

closed 2026-11-06 by leftger