embassy-rs/embassy #3818: NonInvertingPin not implemented for OPAMP pins

teu5us, 2025-01-28:
The `NonInvertingPin` impls seem to be generated based on metapac [here](https://github.com/embassy-rs/embassy/blob/f6532e8f2cd20ed2e63d5c68c2027e959eec30d2/embassy-stm32/build.rs#L1231) for pins with `VP*` signals, however, those seem to be `VINM` signals according to manual (stm32h743vi). Is there anything I can do about in?

teu5us, 2025-01-28:
I'm not sure about the common naming as I'm new to this. VINP/VINM/VOUT feels right to me for some reason. `InvertingPin` is also not implemented, and I cannot find anything doing that in the sources as apposed to output pins and inverting pins. Besides, `buffer_ext` only takes `NonInvertingPin` as input where I can see inverting pin used in the C code that I have. So is there anything I can do right now to make this work?

Dirbaio, 2025-01-28:
for inverting pins you'd have to duplicate this to make the impls 

https://github.com/embassy-rs/embassy/blob/f6532e8f2cd20ed2e63d5c68c2027e959eec30d2/embassy-stm32/build.rs#L1231-L1239

and add new constructors to `struct Opamp` for use cases that use inverting pins.

teu5us, 2025-01-28:
G4 reference manual lists non-inverted inputs as VINP, I guess that it is the same for F3. I think that the naming should be kept as per the manual

[image]

QIANWC, 2025-03-02:
> I'm not sure about the common naming as I'm new to this. VINP/VINM/VOUT feels right to me for some reason. `InvertingPin` is also not implemented, and I cannot find anything doing that in the [...]

Both inverting pin and opamp standalone mode are not implemented, opamp.rs need to add:
macro_rules! impl_opamp_vn_pin
pub fn standalone(vp, vn, vout)