embassy-rs/embassy #3835: esp32s3 embassy-executor v0.7 missing main macro

rpisarew, 2025-02-02:
I have some issues to find the right feature flags for a `esp32s3` to get the v0.7 executor running.
I tried all of the possible feature flags, but I don't get it to build.. Any advice what I am doing wrong?

Cargo.toml
```toml
[package]
name = "vapo-control"
version = "0.1.0"
edition = "2021"
license = "MIT OR Apache-2.0"

[dependencies]
esp-backtrace = { version = "0.15", features = [
  "esp32s3",
  "exception-handler",
  "panic-handler",
  "println",
] }
esp-hal = { version = "0.23", features = ["esp32s3", "unstable"] }
esp-println = { version = "0.13", features = ["esp32s3", "log"] }
log = { version = "0.4" }
esp-alloc = { version = "0.6" }
embassy-executor = { version = "0.7", features = [
  "nightly",
  "executor-thread",
] }
esp-hal-embassy = { version = "0.6", features = ["esp32s3"] }
```

main.rs
```rust
#![no_std]
#![no_main]
#![feature(impl_trait_in_assoc_type)]

use embassy_executor::Spawner;
use esp_backtrace as _;
use esp_hal::delay::Delay;
use esp_hal::time::Duration;
use esp_hal::timer::timg::TimerGroup;

extern crate alloc;
use core::mem::MaybeUninit;

fn init_heap() {
    const HEAP_SIZE: usize = 32 * 1024;
    static mut HEAP: MaybeUninit<[u8; HEAP_SIZE]> = MaybeUninit::uninit();

    unsafe {
        esp_alloc::HEAP.add_region(esp_alloc::HeapRegion::new(
            HEAP.as_mut_ptr() as *mut u8,
            HEAP_SIZE,
            esp_alloc::MemoryCapability::Internal.into(),
        ));
    }
}

#[embassy_executor::main]
async fn main(spawner: Spawner) -> ! {
    init_heap();

    let peripherals = esp_hal::init(esp_hal::Config::default());
    let delay = Delay::new();
    let timg0 = TimerGroup::new(peripherals.TIMG0);

    esp_hal_embassy::init(timg0.timer0);
    esp_println::logger::init_logger_from_env();

    loop {
        log::info!("Hello world!");
        delay.delay(Duration::millis(500));
    }
}
```

lulf, 2025-02-03:
You should use the macros from esp-hal, like https://github.com/esp-rs/esp-hal/blob/main/examples/src/bin/embassy_hello_world.rs

rpisarew, 2025-02-03:
> You should use the macros from esp-hal, like https://github.com/esp-rs/esp-hal/blob/main/examples/src/bin/embassy_hello_world.rs

Thank you very much :)

closed 2025-02-03 by rpisarew