embassy-rs/embassy #3839: [RP235x] various hangs when using multicore

wowkster, 2025-02-03:
I've observed various different hangs when attempting to use core 1 on both the RP2350A and RP2350B. Specifically I've noticed async SPI on core 1 will hang (the core 1 executor continues to run but the task awaiting the SPI future will never be woken up), and channels will hang on the sender side when sending from core 0 to core 1. Neither of these happen 100% of the time but enough that it makes it almost impossible to use the second core reliably.

I've written an [minimal reproducible example](https://github.com/adom-inc/embassy/blob/cee99c9cb6c9ea4a4492616e8fcc2c93a7b590c4/examples/rp23/src/bin/multicore_hang_demo.rs) that works on a bare Pico 2 from my testing. Given the nature of the bug, it's hard to tell if it could be related to the critical section implementation, the executor implementation, or something else entirely.

Dirbaio, 2025-02-04:
seems to be something with cross-core wakes. 

This hangs pretty reliably: https://gist.github.com/Dirbaio/c04beb4d5e08a8508f430227cf626802

We haven't been able to repro on rp2040 yet, so it's unclear if it's a bug new to the rp235x

jamesmunns, 2025-02-04:
Might be worth trying to disable `timer.dbgpause` on any relevant timers, it's possible that the debug core logic has changed between both parts, if you are using RTT (and scanning the ring buffer pointers) it might register the debug core as "active"?

https://forums.raspberrypi.com/viewtopic.php?t=380245

PR #3851 (2025-02-05): rp235x: fix atomics, fix SIO spinlock stuck bug, fix missing core1 reset.
- **rp: Workaround "SIO spinlock stuck bug", reset PROC1 at boot.**
- **rp: make atomics work properly between cores in rp235x.**

closes #3839
closed 2025-02-05 by Dirbaio