embassy-rs/embassy #3847: embassy-time feature "tick-hz-1_000" conflicts with embassy-rp feature "time-driver"

mschnell1, 2025-02-04:
I did a replacement for `embassy_rp/time_driver.rs` that allows to use `TIMER0`, `Timer1 `and the `Powman AON Timer` (with the Low Power Oscillator) 
The last uses a hardware tick with 1 KHz.

Hence I tried to configure the system Tick rate by the feature `"tick-hz-1_000"` so that the tick rate is that of the hardware timer and does not need to be multiplied / divided when setting the alarm or calling "now". 

The (local source) time driver works fine with setting `"tick-hz-1_000"` and without (i.e. using 1 MHz ticks) when using the AON (1Khz) timer. 

But when using the standard driver (by  embassy-rp feature "time-driver"), it seemingly is not allowed to change the system Tick rate.

Am I doing anything wrong ? 

Also see:
https://github.com/embassy-rs/embassy/issues/2961

Dirbaio, 2025-02-04:
The `time-driver` feature forces `tick-hz-1_000_000` because TIMER0 ticks at a fixed 1Mhz. 

if you want to add support for the always-on timer as an alternate time driver, it'd make sense to change `embassy-rp` to have multiple cargo features. 

`time-driver-timer0`, `time-driver-timer1` which enables `tick-hz-1_000_000` 
`time-driver-aot` which enables `tick-hz-1_000`

This way the user chooses the wanted timer, and the tick rate is automatically configured correctly for it.

mschnell1, 2025-02-04:
Yep. That makes sense !

Doing tests with `TIMER1`, I found that I easily could make it run at certain Rates  via the `Ticks` Generator, but 1 Khz is way too slow, as the Generator only allows dividing by 511. 

(In fact we could easily have rates ... 32_483 ...12_000_000 with appropriate configuration. But that do3es not make much sense.)

I do agree that very low power projects, that want to use the LPOSC ad the AON timer will be fine with a fixed predefined Tick rate of 1_000 while other projects will prefer using Timer0 (or maybe Timer1) and be fine with a fixed predefined tick rate of 1_000_000. 

Thanks for listening !

Dirbaio, 2025-02-04:
oh there's a divider, I didn't know that.

If we wanted to support that, the `time-driver-timer1` feature could not set any tick rate, and instead *read* the tick rate the user has selected and configure the timer with the best divider possible. Maybe check if the obtained freq is within x% of the desired freq and panic if not.

embassy-stm32's time driver works this way.

Dirbaio, 2025-02-04:
would you consider sending a PR for the always-on time driver? that'd be awesome 👀

mschnell1, 2025-02-04:
Maybe do a fork and commit ?

closed 2025-02-04 by mschnell1

mschnell1, 2025-02-05:
@Dirbaio done 
-> https://github.com/mschnell1/embassy

-> https://github.com/mschnell1/embassy/blob/rp-timer-driver-with-time-driver-timer1-and-time-driver-aot/embassy-rp/src/time_driver.rs

please review 
Thanks

mschnell1, 2025-02-11:
@Dirbaio 

I did do a pull request, but I can't get it through the pipeline. 

can you take a look ?

mschnell1, 2025-04-04:
@Dirbaio 
For the current project we decided to use an STM32 instead of the RP2350.
Hence I will not "professionally" work on the RP3250 time driver implementation.
But I do think my code is decently workable and just needs some beatifying, integrating (configs) and testing. I am really convinced that using the "Platform Timer" should be the default with the RP235x and the "AON Timer" should be eligible for low power projects. 
Hence I would be inclined to help to maker this happen.
Please let me know if this might be planned with the Embassy RP library.

mschnell1, 2026-08-16:
BTW.: 
IMHO we should change the behavior of `tick-hz-?????`in a way that the user does not get erroneous behavior when using other values than `tick-hz-1_000_000`. 
At least we could check the correct setting in time_driver.rs (by `static assert!()`). With a different timer but TIMER0 we could use different resolutions without compromising backwards compatibility.-