embassy-rs/embassy #3854: RP 2x chips: Embassy Executor and two cores

mschnell1, 2025-02-06:
Just some questions: 
- Is it allowed and does it make sense (and when ?), to run one (or more) Executors on both cores ?
- Interrupts can be assigned to either Core,  how does this affect the Executor(s) (preemption, etc) ?
- There is only a single timer driver. Same obviously uses a hardware timer (Timer 0 by default on an RP235x, the 2040 only has a single timer) the interrupt of which is assigned to one core (supposedly core 0). Is it no problem to use that with an executor running on the other core (supposedly it will not do preemption) ? With a 235x,. would it make sense to use a second timer for the executor on the other core ?

Dirbaio, 2025-02-06:
> Is it allowed and does it make sense (and when ?), to run one (or more) Executors on both cores ?

One per core is supported and works fine. See the [example](https://github.com/embassy-rs/embassy/blob/main/examples/rp23/src/bin/multicore.rs).

InterruptExecutor doesn't currently work in multicore, see [issue](https://github.com/embassy-rs/embassy/issues/1634). so you can't create multiple per core yet.

> Interrupts can be assigned to either Core, how does this affect the Executor(s) (preemption, etc) ?

It causes InterruptExecutor to not work yep, see the above issue.

> There is only a single timer driver. (...) Is it no problem to use that with an executor running on the other core (supposedly it will not do preemption) ?

it's not a problem. The time driver runs in core0 and handles wakes for tasks from both cores.

> With a 235x,. would it make sense to use a second timer (or a second alarm) for the executor on the other core ?

I'd say no, or at least not for now. It's possible to make a time driver that has two timer queues one for each core, so all the timer wakes stay "local" to the core. It might give a small perf boost but the increase in complexity is big.

mschnell1, 2025-02-06:
Do I understand correctly that for allowing for preemption (of a core) the interrupt executor is necessary ? 
Do I understand correctly that I can have multiple "standard" executors on each core (providing cooperative multitasking)  ? 
Do I understand correctly that additionally multiple interrupt executor (allowing for preemption) on one core are allowed, but then no interrupt executor on the other ? 
Hence trying to use a second timer for the other core supposedly would need to solve https://github.com/embassy-rs/embassy/issues/1634 first, which does not have much priority. 

Hence we will stay with no preemption on the second core, which will be perfectly fine with close top all projects.

Dirbaio, 2025-02-06:
> Do I understand correctly that for allowing for preemption (of a core) the interrupt executor is necessary ?
Yes

> Do I understand correctly that I can have multiple "standard" executors on each core (providing cooperative multitasking) ?
No, you can only have one per core. You can spawn multiple tasks in it though, and they'll get cooperatively scheduled.

> Do I understand correctly that additionally multiple interrupt executor (allowing for preemption) on one core are allowed, but then no interrupt executor on the other ?
No, due to #1634 InterruptExecutor is currently broken even if you use it only in 1 core.

> Do I understand correctly that for a time based preemption I always would need a second alarm (-interrupt) and an interrupt executor ? (Or even better user the second core (without an executor) for fully parallel work, which will make sense in close to all projects.)

The concept of "alarm" doesn't exist since embassy-time 0.4, embassy-time-driver 0.2. Now there's a single queue for all cores, all priorities.

> Maybe this even should be the default with RD235x ?!?!?

it's riscv-only, so probably not

mschnell1, 2025-02-06:
The RD235x documentation says (e.g. on pare 37): 
"The RISC-V standard platform timer (MTIME, MTIMEH), which is also usable by Arm processors, "

mschnell1, 2025-02-06:
> oh wow that's very interesting, didn't see that. thanks!

Do you think I should try it and update my driver code ?

mschnell1, 2025-02-10:
The code for the Risk V System Timer seems to work. 
I suppose you will want to do some modifications with the "feature" configuration (as I introduced  feature = `"time-driver-timer0"` which would be sensible if the RiskV timer gets the default.
@Dirbaio  Please review !
Thanks !

mschnell1, 2025-02-10:
> Hmm I did test the `multiprio.rs` example from `\embassy\examples\rp\src\bin\`, which uses the `InterruptExecutor `and seems to work fine.

@Dirbaio: any comment ?

Dirbaio, 2025-02-10:
> @Dirbaio Please review !

review what? you haven't opened any pull request

> Hmm I did test the multiprio.rs example from \embassy\examples\rp\src\bin\, which uses the InterruptExecutor and seems to work fine.

The fact that it seems to work fine doesn't mean it will work fine in all cases. The issue I linked explains what the problem is.

mschnell1, 2025-02-10:
I did not suppose I can write to this repository, do I did a fork: https://github.com/mschnell1/embassy