embassy-rs/embassy #3855: Hard to realize embassy-boot on STM32F401xB/C?

dingari, 2025-02-06:
I've been trying to incorporate `embassy-boot` into my project, which runs on STM32F401. We're currently working with the xC/xE variants (256/512K flash).

Given that the `boot_loader` module works on a `page_size` that is the larger one of `ACTIVE::ERASE_SIZE` and `DFU::ERASE_SIZE`, and the fact that `DFU` must be larger than `ACTIVE` by at least one page, this leaves very few options for lower-memory variants.

The `ERASE_SIZE` constants are determined by the sector size for each region of the flash. For STM32F401xC, the regions are 

* Region 1: 0x0800 0000 - 0x0800 FFFF  (64K / 4 sectors)
* Region 2: 0x0801 0000 - 0x0801 FFFF (64K / 1 sector)
* Region 3: 0x0802 0000 - 0x0803 FFFF (128K / 1 sector)

The most straightforward is to put BOOTLOADER and BOOTLOADER_STATE in region 1, the ACTIVE in region 2, and DFU in region 3 (I'm very much assuming that a partition can't overlap region boundaries). This leaves `page_size = 128K`, and immediately violates the booloader's assertions:

```
assert_eq!(active.capacity() as u32 % page_size, 0);
assert!(dfu.capacity() as u32 - active.capacity() as u32 >= page_size);
```

I haven't figured out a sensible way of arranging the memory for this to make sense. Am I missing something obvious, or is the bootloader lib just not suitable for this flash memory structure?

T xE variant (512K flash), adds two 128K sectors to the flash, so I guess I would place the DFU in two of those, and the ACTIVE in one. That'll leave 128K of actual "usable" flash for the application code.

(is this maybe the reason that `embassy-boot` has examples for f3 and f7, but not f4?)

Dirbaio, 2025-02-06:
> is the bootloader lib just not suitable for this flash memory structure?

yep that's unfortunately correct, with just 1x 128kb sector you can't use embassy-boot.

> T xE variant (512K flash), adds two 128K sectors to the flash, so I guess I would place the DFU in two of those, and the ACTIVE in one. That'll leave 128K of actual "usable" flash for the application code.

That's correct as well, it's a massive waste. Blame ST for picking such huge sectors! :( 

There's two things you can do:
- Pick another stm32 familiy that has smaller sectors
- Do not use embassy-boot, use another bootloader (or make one yourself) that doesn't do A/B partition updates, instead the bootloader itself does the download and overwrites the application. Depending on the kind of device you're building it might be okay. There's no "librarified" bootloader that does this for Embassy, but doing your own is not that hard. See this for example: https://github.com/embassy-rs/eprobe/blob/main/bootloader/src/main.rs

lulf, 2025-02-13:
Closing this as I think you've got an answer @dingari, please reopen if not.

closed 2025-02-13 by lulf