embassy-rs/embassy #3856: Embassy-time has multiple versions of embedded_hal dependencies that causes conflicts on traits

LelouBil, 2025-02-06:
I am having the error below because embassy_time pulls the wrong version of embedded_hal
```rust
error[E0277]: the trait bound `GpioPin<5>: embedded_hal::digital::OutputPin` is not satisfied because the trait comes from a different crate version
   --> src\init.rs:40:9
    |
40  |         led: peripherals.GPIO5,
    |         ^^^^^^^^^^^^^^^^^^^^^^ the trait `embedded_hal::digital::OutputPin` is not implemented for `GpioPin<5>`
    |
note: there are multiple different versions of crate `embedded_hal` in the dependency graph
   --> embedded-hal-1.0.0\src\digital.rs:124:1
    |
124 | pub trait OutputPin: ErrorType {
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ this is the required trait
    |
   ::: src\entry.rs:4:5
    |
4   | use embedded_hal::digital::OutputPin;
    |     ------------ one version of crate `embedded_hal` is used here, as a direct dependency of the current crate
    |
   ::: src\init.rs:6:5
    |
6   | use embassy_time::{Duration, Timer};
    |     ------------ one version of crate `embedded_hal` is used here, as a dependency of crate `embassy_time`
    |
   ::: esp-hal-0.23.1\src\gpio\mod.rs:649:1
    |
649 | pub struct GpioPin<const GPIONUM: u8>;
    | ------------------------------------- this type doesn't implement the required trait
    |
   ::: embedded-hal-0.2.7\src\adc.rs:39:1
    |
39  | pub trait Channel<ADC> {
    | ---------------------- this is the found trait
    = help: you can use `cargo tree` to explore your dependency tree

```

I think a possible fix would be to separate the two embedded_hal versions with feature flags

Dirbaio, 2025-02-07:
`GpioPin<5>` doesn't implement OutputPin. the [`Output` driver](https://docs.esp-rs.org/esp-hal/esp-hal/0.23.1/esp32c3/esp_hal/gpio/struct.Output.html) does. Try doing `Output::new(peripherals.GPIO5)`.

The error you're seeing is a bug in rustc. It correctly points out that `GpioPin` doesn't implement `OutputPin`, but then it goes wildly off the rails and points angrily to embassy-time accusing it of pulling in embedded-hal 0.2 for no reason, it has nothing to do with it. And at the bottom it even says "this is the found trait" pointing to an ADC trait which reaaally has nothing to do with this. What.

LelouBil, 2025-02-07:
Okay yeah I messed up, using the OutputPin Trait from esp-hal directly made the problem go away.

Weird compiler error still.

Thanks for pointing that out.

closed 2025-02-07 by LelouBil