embassy-rs/embassy #3874: Type system related to BufferedCan is unsound

blaa, 2025-02-12:
Tested on embassy f09277b2a37cb77d3adf6675146c282fbcb9c955

Problem: Types allow to drop and deinitialize (?) CAN while holding a broken reader/writer types.

My ultimate goal: I want to send CAN frames and not BLOCK if CAN fails and receives no Acks from other devices. Normal interface available via can.split() doesn't seem to publish any len/try/wouldblock methods. BufferedCan does though.

My simplified CAN setup code:
```
static TX_BUF: StaticCell<can::TxBuf<4>> = StaticCell::new();   
static RX_BUF: StaticCell<can::RxBuf<4>> = StaticCell::new(); 
// I only keep this around so that CAN keeps working. 
static BUFFERED_CAN: StaticCell<embassy_stm32::can::BufferedCan<'static, 4, 4>> = StaticCell::new();

// (...)

let can = can::CanConfigurator::new(p.FDCAN1, p.PB8, p.PB9, CanIrqs);
can.properties().set_extended_filter(  
    can::filter::ExtendedFilterSlot::_0, 
    can::filter::ExtendedFilter::accept_all_into_fifo1(),
);  
can.set_bitrate(250_000); 
let can = can.start(can::OperatingMode::NormalOperationMode); 

let tx_buf = TX_BUF.init(can::TxBuf::<4>::new());
let rx_buf = RX_BUF.init(can::RxBuf::<4>::new());

let buffered = can.buffered(tx_buf, rx_buf);
let writer = buffered.writer();  
let reader = buffered.reader();  
BUFFERED_CAN.init(buffered); // <- This is required, but not enforced
   
Self { 
    can_tx: Mutex::new(writer),  
    can_rx: reader,  
}
```

I think problem is that buffered writer()/reader() don't borrow from BufferedCan, relevant part of fdcan.rs:480:
```
/// Returns a sender that can be used for sending CAN frames.                          
pub fn writer(&self) -> BufferedCanSender {                                 
    BufferedCanSender {                                               
        tx_buf: self.tx_buf.sender().into(),                                  
        waker: self.info.tx_waker,                                               
    }                                                                   
}                                                                            

/// Returns a receiver that can be used for receiving CAN frames. Note, each CAN frame will only be received by one receiver.
pub fn reader(&self) -> BufferedCanReceiver {        
    self.rx_buf.receiver().into()              
}
```

I believe either documenting this behaviour (albeit not really Rusty) or changing the lifetime dependency would be a solution.

blaa, 2025-02-12:
I haven't thoroughly checked but https://github.com/embassy-rs/embassy/issues/3610 maybe related.

vpochapuis, 2025-02-12:
I have also encountered this issue while initializing the BufferedCan driver.

Steps to reproduce:
- Initialize the CAN driver
- Use [`buffered()`](https://docs.embassy.dev/embassy-stm32/git/stm32f439zi/can/struct.Can.html#method.buffered)
- Splitting it into sender and receiver using: [`writer()`](https://docs.embassy.dev/embassy-stm32/git/stm32f439zi/can/struct.BufferedCan.html#method.writer) and [`reader()`](https://docs.embassy.dev/embassy-stm32/git/stm32f439zi/can/struct.BufferedCan.html#method.reader)
- Drop the original driver and only keep the sender and receiver
- Try to send frames over the sender and receive frames from the receiver

The observed behavior is that the buffered receiver will work properly, but the sender will never send frames on the CAN Bus.

Thanks @blaa for finding the origin of the problem and providing a fix!

cschuhen, 2025-03-08:
Hi @blaa, @vpochapuis,

I'm by no means an expert on lifetimes in rust but how about a reference count based fix. Something lilke this:

`
diff --git a/embassy-stm32/src/can/common.rs b/embassy-stm32/src/can/common.rs
index a54b54f6e..7ab53ccd9 100644
--- a/embassy-stm32/src/can/common.rs
+++ b/embassy-stm32/src/can/common.rs
@@ -22,10 +22,10 @@ pub(crate) struct FdBufferedTxInner {
 }
 
 /// Sender that can be used for sending CAN frames.
-#[derive(Copy, Clone)]
 pub struct BufferedCanSender {
     pub(crate) tx_buf: embassy_sync::channel::DynamicSender<'static, Frame>,
     pub(crate) waker: fn(),
+    pub(crate) aliver: fn(bool),
 }
 
 impl BufferedCanSender {
@@ -48,5 +48,20 @@ impl BufferedCanSender {
     }
 }
 
+impl Clone for BufferedCanSender {
+    fn clone(&self) -> Self {
+        defmt::info!("PreClone");
+        (self.aliver)(true);
+        defmt::info!("PostClone");
+        Self{tx_buf: self.tx_buf, waker: self.waker, aliver: self.aliver}
+    }
+}
+
+impl Drop for BufferedCanSender {
+    fn drop(&mut self) {
+        (self.aliver)(false);
+    }
+}
+
 /// Receiver that can be used for receiving CAN frames. Note, each CAN frame will only be received by one receiver.
 pub type BufferedCanReceiver = embassy_sync::channel::DynamicReceiver<'static, Result<Envelope, BusError>>;
diff --git a/embassy-stm32/src/can/fdcan.rs b/embassy-stm32/src/can/fdcan.rs
index c549313f3..4cdbbdf8a 100644
--- a/embassy-stm32/src/can/fdcan.rs
+++ b/embassy-stm32/src/can/fdcan.rs
@@ -478,9 +478,11 @@ impl<'c, 'd, const TX_BUF_SIZE: usize, const RX_BUF_SIZE: usize> BufferedCan<'d,
 
     /// Returns a sender that can be used for sending CAN frames.
     pub fn writer(&self) -> BufferedCanSender {
+        (self.info.tx_aliver)(true);
         BufferedCanSender {
             tx_buf: self.tx_buf.sender().into(),
             waker: self.info.tx_waker,
+            aliver: self.info.tx_aliver,
         }
     }
 
@@ -492,12 +494,12 @@ impl<'c, 'd, const TX_BUF_SIZE: usize, const RX_BUF_SIZE: usize> BufferedCan<'d,
 
 impl<'c, 'd, const TX_BUF_SIZE: usize, const RX_BUF_SIZE: usize> Drop for BufferedCan<'d, TX_BUF_SIZE, RX_BUF_SIZE> {
     fn drop(&mut self) {
+        (self.info.tx_aliver)(false);
         critical_section::with(|_| {
             let state = self.state as *const State;
             unsafe {
                 let mut_state = state as *mut State;
                 (*mut_state).rx_mode = RxMode::NonBuffered(embassy_sync::waitqueue::AtomicWaker::new());
-                (*mut_state).tx_mode = TxMode::NonBuffered(embassy_sync::waitqueue::AtomicWaker::new());
             }
[... 194 lines cut ...]
                    let (ts, rx_frame) = (envelope.ts, envelope.frame);
                    let delta = (ts - last_read_ts).as_millis();
                    last_read_ts = ts;
                    info!(
                        "Rx: {} {:02x} --- {}ms",
                        rx_frame.header().len(),
                        rx_frame.data()[0..rx_frame.header().len() as usize],
                        delta,
                    )
                }
                Err(_err) => error!("Error in frame"),
            }*/

            Timer::after_millis(250).await;
            i = i.wrapping_add(1);
            if i > 5 {
                break;
            }
        }
    info!("Done\n");
}

`

This just does the TX half. Need to do the RX part also. Actually, not quite sure why RX was still working.

cschuhen, 2025-03-08:
Hi @blaa, @vpochapuis,

I'm by no means an expert on lifetimes in rust but how about a reference count based fix. Something lilke this:

Here is a patch (along with how I was testing it) that I think resolves the issue:

https://gist.github.com/cschuhen/7df3aa5762e8f55f5a0b02d30b420595

cschuhen, 2025-03-08:
The only thing I can't figure out is... how was Rx working. It should have been broken also. When the main instance is dropped, both rx_mode and tx_mode should have been put into the NonBuffered mode which kind of makes them do nothing.

blaa, 2025-03-08:
I was considering even a simpler fix (that changes API): making writer/reader simply borrow from buffered (eg. via PhantomData). This of course is way less handy, you must stuff the buffered then somewhere and can't keep it with sender/writer in the same struct (as that would be self-referential). So most probably another static variable.

So in terms of practicality your solution should be better if I understand it right. Dunno if simple borrow wouldn't match Embassy API just fine.

PR #3956 (2025-03-11): STM32 CAN: Count senders and receivers so that we don't close down early.
This is intended to fix these issues:
Fixes #3874 
Fixes #3610

Store a count of the number of senders and receivers in the state so that we only disable the buffered mode when the last one is dropped.