embassy-rs/embassy #3883: Broken embedded_hal and embedded_hal_nb reads for stm32 BufferedUartRx

DrTobe, 2025-02-14:
We (with @embediver) were using the `BufferedUartRx` from `embassy-stm32` via the `embedded_hal_nb` traits but it did not work as expected. But the `embedded_io::Read` implementation works.

Comparing [`embedded_hal_02::serial::Read for BufferedUartRx`](https://github.com/embassy-rs/embassy/blob/29d05328f9b33953b8476ca3a5a7b1e3c9adeaf2/embassy-stm32/src/usart/buffered.rs#L910) (which is also used for `embedded_hal_nb`) with [`BufferedUartRx::blocking_read`](https://github.com/embassy-rs/embassy/blob/29d05328f9b33953b8476ca3a5a7b1e3c9adeaf2/embassy-stm32/src/usart/buffered.rs#L576) (which is used for `embedded_io::Read`) reveals that the implementations are completely different. The working `blocking_read` implementation uses the underlying ring buffer while the broken `embedded_hal_02::serial::Read` implementation seems to access the registers directly.

I assume this is some kind of copy-paste-error which may be relatively easy to fix. Could someone confirm my assumption? If so, should we supply a fix ourselves (which would need some work to understand the code properly first) or is someone able to quickly do it?

Dirbaio, 2025-02-14:
Oh yes, that's very wrong. The `nb` implementation should also pop bytes from the ringbuffer, not from the hardware directly.

embedded-rust-iml-user, 2025-02-17:
> If so, should we supply a fix ourselves (which would need some work to understand the code properly first) or is someone able to quickly do it?

If someone wants to step up, feel free. Otherwise, we do our best as soon as we find the time to do it :) Just please let us know then so we don't do the work twice.

DrTobe, 2025-02-17:
> If so, should we supply a fix ourselves (which would need some work to understand the code properly first) or is someone able to quickly do it?

If someone wants to step up, feel free. Otherwise, we do our best as soon as we find the time to do it :) Just please let us know then so we don't do the work twice.

kpfleming, 2025-05-23:
This should be closed as #3915 was merged.

closed 2025-05-23 by Dirbaio