embassy-rs/embassy #3900: STM32U5: uart.read_until_idle() every second transfer also returns the prevoius data + new data.

vDorst, 2025-02-18:
I make a repo to easy demostrate the issue: https://github.com/vDorst/uartdmabu

Setup: STM32 Development board (STM32H573I-DK) with STM32H573IIK3Q.
Connect USB cable to STLINK port. Used for debug (probe-rs) and VCP (virtual comport).
I used putty to connect VCP port with settings: 115200-8-n-1.
UART7 can also be used. In my project that was I first noticed this behaviour.

After reset, you see 'Hello!' in putty terminal.

If you slowly (1 key/sec) type `1` to `0` in putty you can observe this behavior

Putty ouput
```
Hello!

Got: 1
Got: 2
Got: 23
Got: 4
Got: 45
Got: 6
Got: 67
Got: 8
Got: 89
Got: 0
```

Probe-rs output:
```
# cargo run --release
   Compiling uartdmabug v0.1.0 (<snip>\\uartdmabug)
    Finished `release` profile [optimized + debuginfo] target(s) in 2.45s
     Running `probe-rs run --chip STM32H573IIKx --connect-under-reset target\thumbv8m.main-none-eabihf\release\uartdmabug`
      Erasing ✔ 100% [####################]  32.00 KiB @ 471.92 KiB/s (took 0s)
  Programming ✔ 100% [####################]  28.00 KiB @  82.02 KiB/s (took 0s)                                                                                                                                                                                        
     Finished in 0.41s
Start Init
0.000091 ERROR UART7: Zero size!
└─ uartdmabug::__mb_link_task_task::{async_fn#0} @ src\main.rs:43
0.000091 ERROR UART1: Zero size!
└─ uartdmabug::__mb_link_task_task::{async_fn#0} @ src\main.rs:43
UART1: Got: n_bytes 1 data: [31]
UART1: Got: n_bytes 1 data: [32]
UART1: Got: n_bytes 2 data: [32, 33]
UART1: Got: n_bytes 1 data: [34]
UART1: Got: n_bytes 2 data: [34, 35]
UART1: Got: n_bytes 1 data: [36]
UART1: Got: n_bytes 2 data: [36, 37]
UART1: Got: n_bytes 1 data: [38]
UART1: Got: n_bytes 2 data: [38, 39]
UART1: Got: n_bytes 1 data: [30]
```

It is not only a single character but also has the same behavior with larger data.

If I parse `ACBD`, `EFGH`, `IJLK` as seperet groups into putty I got this output

Putty ouput
```
Got: ACBD
Got: ACBDEFGH
Got: IJLK
```

Probe-rs output:
```
UART1: Got: n_bytes 4 data: [41, 43, 42, 44]
UART1: Got: n_bytes 8 data: [41, 43, 42, 44, 45, 46, 47, 48]
UART1: Got: n_bytes 4 data: [49, 4a, 4c, 4b]
```

I also tested this code on stm32u0 (arm thumbv6) and work fine.
But stm32u0 has a different DMA block.
So I think there is something wrong with the DMA setup.

I also noticed that:
- the first `read_until_idle()` always return Ok(0).
- First two transfers seems to be fine.

@Dirbaio you seems the initial author of the stm32u5 dma code. 
Any idea?

Dirbaio, 2025-02-18:
repo link is 404

vDorst, 2025-02-18:
Fixed, was missing a g at the end

foxxorcat, 2025-11-21:
I'm encountering the same issue when using the `STM32F103`. I applied the [#a8d31f ](https://github.com/fan2nd/embassy/commit/a8d31ff8ce658812d535b3927d350eb4b782617f) patch, but it only delayed the problem by 2 minutes. I also tried using `BufferedUart`, but it's still affected by this bug—returning the truncated byte from the previous transfer only when new data arrives. Currently, my temporary workaround is to recreate the USART instance.

log output:
```
5.650428 Rx: L=8 D=[1,3,0,0,0,6,c5,c8]
...
182.550384 Rx: L=8 D=[1,3,0,0,0,6,c5,c8]
183.561767 Rx: L=7 D=[1,3,0,0,0,6,c5]
184.585357 Rx: L=8 D=[c8,1,3,0,0,0,6,c5]
185.591827 Rx: L=8 D=[c8,1,3,0,0,0,6,c5]
...
``` 
The subsequent behavior follows a consistent pattern, with a one-byte shift occurring approximately every 30 seconds (when the patch is not applied).

usbalbin, 2025-12-03:
I hope #4981 is it