embassy-rs/embassy #3904: embassy-net: Thread safety issue?

showier-drastic, 2025-02-20:
Hi,

I'm occasionally getting this error when trying to send UDP packets from an interrupt (by directly calling `poll_send_to` of `UdpSocket`):

```
18.739376 ERROR panic-probe-0.3.2/src/lib.rs:104  panicked at embassy/embassy-net/src/lib.rs:388:27:
already borrowed: BorrowMutError
```

The fact that you can create such crash without any unsafe code, IMHO, means that there's a potential unsoundness:

The `Stack` of embassy-net struct directly use `RefCell` rather than `Mutex` to achieve shared access. As such, you can only operate all network-related things within one executor.

Is this correct?

Dirbaio, 2025-02-20:
> trying to send UDP packets from an interrupt

how are you getting the `Stack` from main to interrupt? [It's not Send](https://docs.embassy.dev/embassy-net/git/default/struct.Stack.html#impl-Send-for-Stack%3C'd%3E), so you shouldn't have been able to. Have you used any `unsafe`?

> The Stack of embassy-net struct directly use RefCell rather than Mutex to achieve shared access. As such, you can only operate all network-related things within one executor.

Yes, that's the intended behavior. The RefCell makes it not be `Send`, so you should've been prevented at compile time from sharing it between main and interrupt.

showier-drastic, 2025-02-20:
> how are you getting the Stack from main to interrupt? [It's not Send](https://docs.embassy.dev/embassy-net/git/default/struct.Stack.html#impl-Send-for-Stack%3C'd%3E), so you shouldn't have been able to. Have you used any unsafe?

No, I haven't used any unsafe. But I used `OnceLock`. Maybe the problem is the `OnceLock`?

```rust
use embassy_sync::once_lock::OnceLock;

static UDP: OnceLock<udp::UdpSocket<'static>> = OnceLock::new();

#[embassy_executor::task]
async fn net_task(mut runner: embassy_net::Runner<'static, Device>) -> ! {
    runner.run().await
}

fn send_udp() {
    if let Some(udp) = UDP.try_get() {
        udp.poll_send_to(/* ... */);
    }
}

#[interrupt]
fn SOME_INTERRUPT() {
    send_udp();
}

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let eth = eth::Ethernet::new( /* ... */ );

    // Init network stack
    static RESOURCES: StaticCell<embassy_net::StackResources<3>> = StaticCell::new();
    let (stack, runner) = embassy_net::new( /* ... */ );

    // Launch network task
    defmt::unwrap!(spawner.spawn(net_task(runner)));

    // Ensure DHCP configuration is up before trying connect
    stack.wait_config_up().await;

    defmt::info!("Network task initialized");

    static RX_META: StaticCell<[udp::PacketMetadata; 16]> = StaticCell::new();
    static RX_BUFFER: StaticCell<[u8; 4096]> = StaticCell::new();
    static TX_META: StaticCell<[udp::PacketMetadata; 16]> = StaticCell::new();
    static TX_BUFFER: StaticCell<[u8; 4096]> = StaticCell::new();

    let mut socket = udp::UdpSocket::new(
        stack,
        RX_META.init([udp::PacketMetadata::EMPTY; 16]),
        RX_BUFFER.init([0; 4096]),
        TX_META.init([udp::PacketMetadata::EMPTY; 16]),
        TX_BUFFER.init([0; 4096]),
    );
    socket.bind(9400).unwrap();

    UDP.init(socket).map_err(|_| ()).unwrap();

    loop { /* ... */ }
}

```

Dirbaio, 2025-02-20:
oh yep. this should have `where T: Sync`

https://github.com/embassy-rs/embassy/blob/2e7a2b61275d70d7d09cb676490e2cba4d2bde3f/embassy-sync/src/once_lock.rs#L45

wmmc88, 2025-06-18:
> oh yep. this should have `where T: Sync`
> 
> [embassy/embassy-sync/src/once_lock.rs](https://github.com/embassy-rs/embassy/blob/2e7a2b61275d70d7d09cb676490e2cba4d2bde3f/embassy-sync/src/once_lock.rs#L45)
> 
> Line 45 in [2e7a2b6](/embassy-rs/embassy/commit/2e7a2b61275d70d7d09cb676490e2cba4d2bde3f)
> 
>  unsafe impl<T> Sync for OnceLock<T> {}

@Dirbaio I've fixed this in #4320