embassy-rs/embassy #3905: Example uart_async.rs may trigger assertion

mdelete, 2025-02-20:
A lot of sample code shows the usage of _UART_ in combination with _DMA_. For example _examples/stm32wl/src/bin/uart_async.rs_ has some code that creates a simple uart echo service:

    let mut usart2 = Uart::new(p.LPUART1, p.PC0, p.PC1, Irqs, p.DMA1_CH5, p.DMA1_CH6, config2).unwrap();

    unwrap!(usart1.write(b"Hello Embassy World!\r\n").await);
    unwrap!(usart2.write(b"Hello Embassy World!\r\n").await);

    let mut buf = [0u8; 300];
    loop {
        let result = usart2.read_until_idle(&mut buf).await;
        match result {
            Ok(size) => {
                match usart1.write(&buf[0..size]).await {

I have found that in this particular example _read_until_idle()_ may return _Ok(0)_ which leads to a _write(&buf[0..0])_ which is a zero-length write, which triggers an assertion when having configured DMA:

    panicked at 'assertion failed: len > 0 && len <= 0xFFFF'

in _embassy-rs/embassy/embassy-stm32/src/dma/dma_bdma.rs:685_ 

I can confirm this can be mitigated by using:

    Ok(size) if size > 0 =>

instead.

vDorst, 2025-02-21:
I also have notice this behavior see my issue #3900 .
Do you also see the previous data as described in my issue?
And which chip you are using?

mdelete, 2025-02-21:
@vDorst Your first observation would also be consistent with an electrical issue. STM UARTs are pretty sensitive and may read phantom data, if the transmitter on the other side is not properly driven while idle (by default, the RX line is configured NoPull). But your other observation, where the data gets appended to the buffer, despite already being read, seems to look like a buffer or DMA issue.

I'm working on the **stm32wl55jc-cm4** and cannot confirm this behaviour.

About the Ok(0) issue, I would be fine with that. The call is async and named _read_until_idle_, so I guess it may go idle without having read anything, but then the example would handle this incorrectly...