embassy-rs/embassy #3917: L475 and L476 do not implement BK1NSSPin for any pins

i509VCB, 2025-02-25:
L475 and L476 have a single bank QSPI which uses bank 1 internally.

However there are no implementors of `BK1NSSPin` for both, which means the QSPI is unusable.
- https://docs.embassy.dev/embassy-stm32/git/stm32l475re/qspi/trait.BK1NSSPin.html
- https://docs.embassy.dev/embassy-stm32/git/stm32l476re/qspi/trait.BK1NSSPin.html

hjeldin, 2025-11-26:
I'm having the same issue. 
I investigated a bit, and it appears that NSS is now named NCS.
I had to edit embassy-stm32/build.rs in:
```diff
        (("quadspi", "BK1_IO0"), quote!(crate::qspi::BK1D0Pin)),
        (("quadspi", "BK1_IO1"), quote!(crate::qspi::BK1D1Pin)),
        (("quadspi", "BK1_IO2"), quote!(crate::qspi::BK1D2Pin)),
        (("quadspi", "BK1_IO3"), quote!(crate::qspi::BK1D3Pin)),
-       (("quadspi", "BK1_NCS"), quote!(crate::qspi::BK1NCSPin)),
+       (("quadspi", "NCS"), quote!(crate::qspi::BK1NCSPin)),
        (("quadspi", "BK2_IO0"), quote!(crate::qspi::BK2D0Pin)),
        (("quadspi", "BK2_IO1"), quote!(crate::qspi::BK2D1Pin)),
        (("quadspi", "BK2_IO2"), quote!(crate::qspi::BK2D2Pin)),
        (("quadspi", "BK2_IO3"), quote!(crate::qspi::BK2D3Pin)),
```

and rename all references to BK1NSSPin in BK1NCSPin in smbassy-stm32/src/qspi/mod.rs
This is because the corresponding [stm32-data-generated json](https://github.com/embassy-rs/stm32-data-generated/blob/main/data/chips/STM32L476RG.json) does not contain the prefix BK1 (nor NSS).
I would create a PR but i don't know if this affects other chips.

Dirbaio, 2025-11-26:
@hjeldin If some chips name it NSS and others NCS, it should be fixed in stm32-data. The name should be consistent in all chips. You can add an `if` somewhere in `stm32-data-gen` to fix the bad name.