embassy-rs/embassy #3918: STM32 H7RS FDCAN Example not working on Nucleo

alvidrezluke, 2025-02-25:
Attempting to run the STM32 H7RS Fdcan example results in the following panic on a Nucleo-H7S3L8.

`
0.000000 ERROR panicked at 'cannot select USBPHYC reference clock with source frequency of Hertz(25000000) Hz, must be one of 16, 19.2, 20, 24, 26, 32 MHz'
└─ embassy_stm32::rcc::_version::init @ /home/alvidrezluke/embassy/embassy-stm32/src/rcc/h.rs:578
0.000000 ERROR panicked at /home/alvidrezluke/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-0.3.10/src/lib.rs:380:5:
explicit panic
`

Changing the HSE frequency to 24_000_000 no longer causes this panic, but no CAN message is ever read either in normal mode, external loopback, or internal loopback mode. On internal loopback and external loopback, after reset TX (PA12) is pulled high and no data is ever transmitted. On normal mode with a SN65HVD230 transceiver, there are no signals either. can.properties().tx_error_count() is also at 0. can.read().await never finishes. Is this a board specific issue or is this an issue with default configs?

kevswims, 2025-03-25:
@alvidrezluke do you have the changes for stm32-data up for this anywhere? I'm trying to get FDCAN working on an H7S and I am also not able to receive a message.

alvidrezluke, 2025-03-25:
@kevswims I'll make a public fork and push up the ADC and FDCAN changes I made to stm32-data. I have tested the CAN at up to 1M baud for Can 2.0 and it seems to work well now. Haven't gotten around to checking the FD Can portion but it looks like it should work.

alvidrezluke, 2025-03-25:
@kevswims here is the stm32-data repo: https://github.com/alvidrezluke/stm32-data-h7rs. The ADC part may require changes to embassy-stm32 so I commented the changes in the perimap out.

kevswims, 2025-03-25:
@alvidrezluke I've got a PR up here: https://github.com/embassy-rs/stm32-data/pull/582 It looks like the H7 R/S use exactly the same CAN peripheral as the H5 and other STM32's instead of the larger one that is in the rest of the H7 line so I have switched it over to that.

closed 2025-03-25 by alvidrezluke