embassy-rs/embassy #3924: Performance Observability

kat-perez, 2025-02-26:
What are the best practices for measuring performance? Are there any tools or crates that can provide event-based diagnostic information like a trace? I'm looking for something that can measure whether tasks are being blocked or not.

lulf, 2025-02-27:
You can enable the trace feature in the executor which will involve some callbacks when scheduling tasks etc. (See the trace rs in the executor source for more info). As an extension of that there is the rtos-trace feature which integrates with System view of youre using that

kat-perez, 2025-03-03:
Thanks! Do you know of any options that are not RTOS-based?

kat-perez, 2025-03-06:
@lulf Could you help walk me through how to use trace.rs? I have enabled it via the feature flag `cargo add embassy-executor --features trace` but i'm getting an "undefined reference" error and I can't find them defined anywhere else in the repo? I'm running the std tick example (`cargo run --bin tick`) with trace feature enabled

/home/X/embassy/embassy-executor/src/raw/trace.rs:17:(.text._ZN16embassy_executor3raw5trace8task_new17h04e7ee5e821644b7E+0x25): undefined reference to `_embassy_trace_task_new'

lulf, 2025-03-06:
You need to define the functions that gets called by the executor, it is quite low level. I can publish some more code on that later, but for now:

```
#[unsafe(no_mangle)]
unsafe extern "Rust" fn _embassy_trace_task_exec_begin(executor: u32, task: u32) {
   
}

#[unsafe(no_mangle)]
unsafe extern "Rust" fn _embassy_trace_task_exec_end(executor: u32, task: u32) {
   
}

#[unsafe(no_mangle)]
unsafe extern "C" fn _embassy_trace_executor_idle(executor: u32) {
   
}

#[unsafe(no_mangle)]
unsafe extern "C" fn _embassy_trace_task_ready_begin(executor: u32, task: u32) {}

#[unsafe(no_mangle)]
unsafe extern "C" fn _embassy_trace_task_new(executor: u32, task: u32) {}
```

So, you can for instance check how long time a task runs by measuring the time between exec_end and exec_begin for a given executor id and task id.

I have some code to generate a trace, but you could also just defmt::info!("{} {} {}", executor, task, Instant::now().as_ticks()); and you should get the ticks when each of these events happen in the log and you can process the information later.

kat-perez, 2025-03-06:
@lulf thanks for clarifying. I'm getting the same undefined symbol error when I enable `rtos-trace` (`undefined symbol: _rtos_trace_system_idle`) Does the RTOS automatically define those functions?

kat-perez, 2025-04-09:
@lulf I tried integrating systemview_target into some sample code I have here (https://github.com/kat-perez/embassy-imxrt/commit/c3a4634247a4514862ccd056f0d2854dd3b2da4d) and SystemView logging shows this, do you have any pointers on what could be going wrong here or how I can debug?

```
11:27:15 - JLink: Device "MIMXRT685S_M33" selected.
11:27:15 - JLink: InitTarget() start
11:27:15 - JLink: InitTarget() end - Took 2.18ms
11:27:15 - JLink: Found SW-DP with ID 0x6BA02477
11:27:15 - JLink: DPIDR: 0x6BA02477
11:27:15 - JLink: CoreSight SoC-400 or earlier
11:27:15 - JLink: Scanning AP map to find all available APs
11:27:15 - JLink: AP[1]: Stopped AP scan as end of AP map has been reached
11:27:15 - JLink: AP[0]: AHB-AP (IDR: 0x84770001, ADDR: 0x00000000)
11:27:15 - JLink: Iterating through AP map to find AHB-AP to use
11:27:15 - JLink: AP[0]: Core found
11:27:15 - JLink: AP[0]: AHB-AP ROM base: 0xE00FE000
11:27:15 - JLink: CPUID register: 0x410FD213. Implementer code: 0x41 (ARM)
11:27:15 - JLink: Feature set: Mainline
11:27:15 - JLink: Cache: No cache
11:27:15 - JLink: Found Cortex-M33 r0p3, Little endian.
11:27:15 - JLink: FPUnit: 8 code (BP) slots and 0 literal slots
11:27:15 - JLink: Security extension: implemented
11:27:15 - JLink: Secure debug: enabled
11:27:15 - JLink: CoreSight components:
11:27:15 - JLink: ROMTbl[0] @ E00FE000
11:27:15 - JLink: [0][0]: E00FF000 CID B105100D PID 000BB4C9 ROM Table
11:27:15 - JLink: ROMTbl[1] @ E00FF000
11:27:15 - JLink: [1][0]: E000E000 CID B105900D PID 000BBD21 DEVARCH 47702A04 DEVTYPE 00 Cortex-M33
11:27:15 - JLink: [1][1]: E0001000 CID B105900D PID 000BBD21 DEVARCH 47701A02 DEVTYPE 00 DWT
11:27:15 - JLink: [1][2]: E0002000 CID B105900D PID 000BBD21 DEVARCH 47701A03 DEVTYPE 00 FPB
11:27:15 - JLink: [1][3]: E0000000 CID B105900D PID 000BBD21 DEVARCH 47701A01 DEVTYPE 43 ITM
11:27:15 - JLink: [1][5]: E0041000 CID B105900D PID 002BBD21 DEVARCH 47724A13 DEVTYPE 13 ETM
11:27:15 - JLink: [1][6]: E0042000 CID B105900D PID 000BBD21 DEVARCH 47701A14 DEVTYPE 14 CSS600-CTI
11:27:15 - JLink: [0][1]: E0040000 CID B105900D PID 000BBD21 DEVARCH 00000000 DEVTYPE 11 TPIU
11:27:35 - ERROR: Error starting recorder.

Check recorder settings and make sure SystemView is enabled on the target.

```

closed 2025-04-21 by kat-perez