embassy-rs/embassy #3934: RP2350 HCI Bluetooth data integrity / clock / speed data copy problem?

brandonros, 2025-02-28:
I could be crazy but according to the `defmt::debug!` logs:

https://github.com/embassy-rs/embassy/blob/17301c00e986c5b8536435ea31ebf5aaf13aed17/cyw43/src/bluetooth.rs#L465

The data I'm sending from the client is not being received by the server in the same `bit` order. It's as if there's a clock issue. It's like the slightest corruption.

```
29.552252 DEBUG HCI rx: [
02, 40, 20, 21, 00, 1d, 00, 04, 00, 52, 22, 00, 06, 

01, 00, 00, 00, <- i do not know why this 01 is here, i am sending 00 00 00 00

00, 00, 07, e5, 
00, 00, 07, ed, 
00, 00, 00, 09, 
66, 69, 6c, 74, 65, 72, 2d, 30, 00
]
```

It's as if this is a problem again? https://github.com/embassy-rs/embassy/pull/1808#issuecomment-1688998970

brandonros, 2025-02-28:
https://github.com/embassy-rs/embassy/pull/3691

I wonder if I'm using the wrong clock divider...

brandonros, 2025-02-28:
There's also a small `btfw` we could sync from upstream I think: https://github.com/georgerobotics/cyw43-driver/commit/7f422fef4ea5bff7285fb78340d3f28f5461cff2

brandonros, 2025-02-28:
```rust
let spi = PioSpi::new(
        &mut pio.common,
        pio.sm0,
        FixedU32::from_bits(0x400), // do not use RM2_CLOCK_DIVIDER or DEFAULT_CLOCK_DIVIDER?
        pio.irq0,
        cs,
        p.PIN_24,
        p.PIN_29,
        p.DMA_CH0,
    );
```

This fixed it for me.

willcrichton, 2025-04-14:
I was struggling to get the `examples/rp235x/src/bin/blinky_wifi.rs` example to run on my Pico 2 W, but this change fixed it for me. Might be worth updating the examples?