embassy-rs/embassy #3936: Issue using published crate versions for Raspberry Pi Pico 2(W)

andrewdavidmackenzie, 2025-03-01:
I am unable to get code (e.g. blinky wifi 2 example) running on a Pi Pico 2 W using the embassy crates published in crates.io.

I tested this using the blinky wifi 2 rp235x example

Using the latest versions from git (i.e. the example as it is), the example is able to run.

Editing the example to use the crate versions from [crates.io](http://crates.io/) (by removing the relative paths for the embassy crates) the example does not run and seems to hang (not produce any blinking or log output) after Erasing/Programming and the "Finished" message.
NOTE:  I had to drop embassy-net from 0.7.0 to 0.6.0 but that shouldn't affect the blinky example.

For reference the edited dependencies:
```
[dependencies]
embassy-embedded-hal = { version = "0.3.0", features = ["defmt"] }
embassy-sync = { version = "0.6.2", features = ["defmt"] }
embassy-executor = { version = "0.7.0", features = ["task-arena-size-98304", "arch-cortex-m", "executor-thread", "executor-interrupt", "defmt"] }
embassy-time = { version = "0.4.0", features = ["defmt", "defmt-timestamp-uptime"] }
embassy-rp = { version = "0.3.0", features = ["defmt", "unstable-pac", "time-driver", "critical-section-impl", "rp235xa", "binary-info"] }
embassy-usb = { version = "0.4.0", features = ["defmt"] }
embassy-net = { version = "0.6.0", features = ["defmt", "tcp", "udp", "raw", "dhcpv4", "medium-ethernet", "dns"] }
embassy-net-wiznet = { version = "0.2.0", features = ["defmt"] }
embassy-futures = { version = "0.1.0" }
embassy-usb-logger = { version = "0.4.0" }
cyw43 = { version = "0.3.0", features = ["defmt", "firmware-logs"] }
cyw43-pio = { version = "0.3.0", features = ["defmt"] }
```

Would it be possible to publish a new version of embassy to crates.io that includes the change (whetever it is) that causes the rp235x example to hang?

BTW: 
I am unable to use the git versions with my own code, as I get a dependency issue with embassy-time-driver via embassy-time between embassy-rp and edge-mdns which I use in my project:

```
error: failed to select a version for `embassy-time-driver`.
    ... required by package `embassy-time v0.4.0`
    ... which satisfies dependency `embassy-time = "^0.4"` of package `edge-mdns v0.5.0`
    ... which satisfies dependency `edge-mdns = "^0.5.0"` of package `porky v0.6.0 (/Users/andrew/workspace/pigg/porky)`
versions that meet the requirements `^0.2` are: 0.2.0

the package `embassy-time-driver` links to the native library `embassy-time`, but it conflicts with a previous package which links to `embassy-time` as well:
package `embassy-time-driver v0.2.0 (https://github.com/embassy-rs/embassy#17301c00)`
    ... which satisfies git dependency `embassy-time-driver` of package `embassy-rp v0.3.0 (https://github.com/embassy-rs/embassy#17301c00)`
    ... which satisfies git dependency `embassy-rp` of package `porky v0.6.0 (/Users/andrew/workspace/pigg/porky)`
Only one package in the dependency graph may specify the same links value. This helps ensure that only one copy of a native library is linked in the final binary. Try to adjust your dependencies so that only one package uses the `links = "embassy-time"` value. For more information, see https://doc.rust-lang.org/cargo/reference/resolver.html#links.

failed to select a version for `embassy-time-driver` which could resolve this conflict

```

9names, 2025-03-01:
Did you add an Image Def to your binary? Those were removed from the examples because the hal will handle that in the next release.

https://github.com/embassy-rs/embassy/pull/3890/commits/6b1706434dfeb80b3659f3d2a1bbe2871ec8ea55#diff-eaaa2bafe1428ccb4df33db3b8aa1caae1bcb22902e9be8f03776c4c49217ee8

In general, if you're using a specific release version of a crate you'll have to look at the examples at the time it was released or you'll get issues like this.

andrewdavidmackenzie, 2025-03-03:
Adding back ImageDef has fixed the examples, using published versions of crates! Thanks!!!

Additional (about trying to reproduce the issue etc)...
I was going to go to the latest release, download the source tarball and repeat my experiment there...but...

It's kind of hard to get all the crates locally, in the state they were when published, as they are all versioned and tagged (https://github.com/embassy-rs/embassy/tags) separately, and and pushed independently (it seems...), and there are no "releases" (https://github.com/embassy-rs/embassy/releases) that push or tag all (and hence no tarball of (embassy-rs).

This work of examples is hard (I know from other projects) I know, and more so for these ones that work with real hardware only (hence hard to CI).

To have them always work with published versions you would have to publish (using relative paths) and then later change the examples to use the published versions. Although I think there maybe workarounds combining versions and paths so it selected the new version if found on crates and the local path version if not?

Either way, thanks for the hint on ImageDef and I will track the next release.

closed 2025-03-03 by andrewdavidmackenzie