embassy-rs/embassy #3940: Timer::after... are very imprecise

s3xysteak, 2025-03-02:
I am using stm32f1 and here is my code for test:
```rs
#![no_std]
#![no_main]

use defmt::*;
use embassy_executor::Spawner;
use embassy_time::{Instant, Timer};
use {defmt_rtt as _, panic_probe as _};

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let _p = embassy_stm32::init(Default::default());

    loop {
        let t1 = Instant::now();
        Timer::after_micros(100).await;
        let t: u64 = t1.elapsed().as_micros();

        info!("{}", t);

        Timer::after_secs(1).await;
    }
}
```
The code as above should ideally output 100 on the console. however my console output is usually around 159. Am I doing something wrong? Or does a more accurate way of timing exist? Sadly, I don't think such an error can be applied to most communication protocols 😢 

And btw here is my Cargo.toml:

<details>

<summary>
Cargo.toml
</summary>

```toml
[package]
name = "stm32f1-embassy"
version = "0.1.0"
edition = "2024"

[dependencies]
embassy-futures = "0.1.0"

# Change stm32f103c8 to your chip name, if necessary.
embassy-stm32 = { version = "0.2.0", features = [
  "defmt",
  "stm32f103c8",
  "unstable-pac",
  "memory-x",
  "time-driver-any",
] }
embassy-sync = { version = "0.6.2", features = ["defmt"] }
embassy-executor = { version = "0.7.0", features = [
  "arch-cortex-m",
  "executor-thread",
  "defmt",
] }
embassy-time = { version = "0.4.0", features = [
  "defmt",
  "defmt-timestamp-uptime",
] }
embassy-usb = { version = "0.4.0", features = ["defmt"] }
defmt = "0.3"
defmt-rtt = "0.4"

cortex-m = { version = "0.7.6", features = [
  "inline-asm",
  "critical-section-single-core",
] }
cortex-m-rt = "0.7.0"
embedded-hal = "0.2.6"
panic-probe = { version = "0.3", features = ["print-defmt"] }
heapless = { version = "0.8", default-features = false }
nb = "1.0.0"
static_cell = "2.0.0"

[profile.dev]
opt-level = "s"

[profile.release]
debug = 2
```
</detail>

I feel very grateful for any help I can get!

s3xysteak, 2025-03-03:
After using a higher tick rate, it was improved. But It still have about 10 us delay with stm32f1(max tick rate). If anyone is looking for a temporary solution, you can use `block_for` instead of `Timer`, which have about 2us delay but will cause block.

Dirbaio, 2025-03-03:
Make sure to follow all the steps here https://embassy.dev/book/#_how_can_i_optimize_the_speed_of_my_embassy_stm32_program

closed 2025-07-08 by Dirbaio