embassy-rs/embassy #3949: No CAN2 reception of frames on STM32F429II

itmmckernan, 2025-03-06:
CAN2 appears to not be receiving frames, in either loopback mode or in normal mode. It is transmitting frames correctly, verified externally.

Modified stm32f4 can example to use CAN2, no frames received. The original example with CAN1 runs correctly
```rust
#![no_std]
#![no_main]

use embassy_executor::Spawner;
use embassy_time::Instant;
use embassy_stm32::{bind_interrupts, can::{filter::Mask32, Can, Fifo, Frame, StandardId}, peripherals::CAN2};
use {defmt_rtt as _, panic_probe as _};
use defmt::{info, unwrap};
use embassy_stm32::can::{Rx0InterruptHandler, Rx1InterruptHandler, SceInterruptHandler, TxInterruptHandler};

bind_interrupts!(struct Irqs {
    CAN2_RX0 => Rx0InterruptHandler<CAN2>;
    CAN2_RX1 => Rx1InterruptHandler<CAN2>;
    CAN2_SCE => SceInterruptHandler<CAN2>;
    CAN2_TX => TxInterruptHandler<CAN2>;
});

#[embassy_executor::main]
async fn main(spawner: Spawner) -> () {
    let p = embassy_stm32::init(Default::default());

    // let mut can = Can::new(p.CAN1, p.PA11, p.PA12, Irqs);
    let mut can = Can::new(p.CAN2, p.PB12, p.PB13, Irqs);
    can.modify_filters().enable_bank(0, Fifo::Fifo0, Mask32::accept_all());
    can.modify_config()
        .set_bitrate(1_000_000)
        .set_loopback(true)
        .set_silent(true);

    can.enable().await;

    let mut i: u8 = 0;
    loop {
        let tx_frame = Frame::new_data(unwrap!(StandardId::new(i as _)), &[i]).unwrap();
        let tx_ts = Instant::now();
        can.write(&tx_frame).await;

        let envelope = can.read().await.unwrap();

        // We can measure loopback latency by using receive timestamp in the `Envelope`.
        // Our frame is ~55 bits long (exlcuding bit stuffing), so at 1mbps loopback delay is at least 55 us.
        // When measured with `tick-hz-1_000_000` actual latency is 80~83 us, giving a combined hardware and software
        // overhead of ~25 us. Note that CPU frequency can greatly affect the result.
        let latency = envelope.ts.saturating_duration_since(tx_ts);

        info!(
            "loopback frame {=u8}, latency: {} us",
            envelope.frame.data()[0],
            latency.as_micros()
        );
        i = i.wrapping_add(1);
    }
}

```

timokroeger, 2025-03-06:
The CAN peripheral has the strange design that the filter banks (which are required for reception) are shared between the two instances. The filter bank is clocked by the CAN1 clock. That means for CAN2 RX to work you also need to enable the CAN1 clock. At some point in the past embassy handled that correctly but maybe it regressed with some RCC refactorings?

itmmckernan, 2025-03-06:
This code appears to work in loopback mode, after disabling the assert_bank_index check in enable for FilterBanks (line 410 in filter.rs)

```rust
#![no_std]
#![no_main]

use embassy_executor::Spawner;
use embassy_time::Instant;
use embassy_stm32::{bind_interrupts, can::{filter::Mask32, Can, Fifo, Frame, StandardId}, peripherals::{CAN1, CAN2}};
use {defmt_rtt as _, panic_probe as _};
use defmt::{info, unwrap};
use embassy_stm32::can::{Rx0InterruptHandler, Rx1InterruptHandler, SceInterruptHandler, TxInterruptHandler};

bind_interrupts!(struct Irqs {
    CAN1_RX0 => Rx0InterruptHandler<CAN1>;
    CAN1_RX1 => Rx1InterruptHandler<CAN1>;
    CAN1_SCE => SceInterruptHandler<CAN1>;
    CAN1_TX => TxInterruptHandler<CAN1>;
    CAN2_RX0 => Rx0InterruptHandler<CAN2>;
    CAN2_RX1 => Rx1InterruptHandler<CAN2>;
    CAN2_SCE => SceInterruptHandler<CAN2>;
    CAN2_TX => TxInterruptHandler<CAN2>;
});

#[embassy_executor::main]
async fn main(spawner: Spawner) -> () {
    let p = embassy_stm32::init(Default::default());

    let mut can1 = Can::new(p.CAN1, p.PA11, p.PA12, Irqs);
    
    can1.modify_filters().enable_bank(15, Fifo::Fifo0, Mask32::accept_all());
    let mut can2 = Can::new(p.CAN2, p.PB12, p.PB13, Irqs);
    
    can2.modify_config()
        .set_bitrate(1_000_000)
        .set_loopback(true) 
        .set_silent(true);

    can2.enable().await;

    let mut i: u8 = 0;
    loop {
        let tx_frame = Frame::new_data(unwrap!(StandardId::new(i as _)), &[i]).unwrap();
        let tx_ts = Instant::now();
        can2.write(&tx_frame).await;

        let envelope = can2.read().await.unwrap();

        let latency = envelope.ts.saturating_duration_since(tx_ts);

        info!(
            "loopback frame {=u8}, latency: {} us",
            envelope.frame.data()[0],
            latency.as_micros()
        );
        i = i.wrapping_add(1);
    }
}

```

itmmckernan, 2025-03-06:
Turns out if I had read the docs, this is properly documented here:
https://docs.embassy.dev/embassy-stm32/git/stm32f429ii/can/struct.Can.html#method.modify_filters
For anyone who stumbles along this thread the proper way of configuring CAN2 banks on bxcan platforms is:
```rust
let mut can1 = Can::new(p.CAN1, p.PA11, p.PA12, Irqs);
// Slave filter Indicies start at 14 on devices with a master and slave CAN interface (28 filter banks)
can1.modify_filters().slave_filters().enable_bank(14, Fifo::Fifo1, Mask32::accept_all());
let mut can2 = Can::new(p.CAN2, p.PB12, p.PB13, Irqs);
```
Closing this as the docs are correct and it is resolved, but being able to call modify_filters on CAN2 and have it do nothing seems potentially unintended.

closed 2025-03-06 by itmmckernan