embassy-rs/embassy #3954: RP235X: Fail after flash with the cargo published version

JJaviMS, 2025-03-08:
Hello everyone, I've detected an error that I could fix but I do not know why it was happening.

I was trying to make an example project with a RP Pico 2 W. The problem that I found was that after flashing the Pico it reboots on mass storage mode after a second of being flashed. Before that reboot there isn't any kind of log which tells what was the problem.

The project copy from the RP235X examples the core files, such as the memory.x the cargo.toml, etc. 

After a lot of try and error I found out that if I flash the program using the published versions of embassy of cargo it fails, but if I change the version tag with a git tag pointing to the embassy repo it works fine.

Here is the not working version of the project: https://github.com/JJaviMS/embassy-temperature-test

And here is the working version: https://github.com/JJaviMS/embassy-temperature-test/tree/working

CBJamo, 2025-03-08:
The version released on crates requires you to create a static ImageDef. Take a look at this older version of the [blinky example](https://github.com/embassy-rs/embassy/blob/e6ce81018bdd1a78d27a285260ac53b276690553/examples/rp23/src/bin/blinky.rs#L16) The git version includes one for you by default. The change happened in #3890 

I recommend using the git version.

9names, 2025-03-08:
Same issue as #3936. I have done most of the prep for a new release of embassy-rp, should push out a new version soon to eliminate this difference between examples and crates.io release

JJaviMS, 2025-03-09:
Thank you very much for the answers now I know which was the problem and I understand this a little bit more.

closed 2025-03-09 by JJaviMS