embassy-rs/embassy #3959: Support for Nordic nRF7001 and nRF7002 wifi ICs?

victorhooi, 2025-03-11:
Is there any chance of Embassy having support for the newer Nordic Wifi chips? (nRF7001 and nRF7002).

You can get [development kits](https://www.nordicsemi.com/Products/Development-hardware/nRF7002-DK) that pair then with a nRF5340 - and I think there's production SOCs starting to be available with that combination as well.

I thought it might be a nice option for a low-powered electronics project I wanted to make, but with wifi support. What do people think?a

lulf, 2025-03-11:
There is an experimental driver here, https://github.com/embassy-rs/nrf70/tree/main

victorhooi, 2025-03-16:
Aha - I had a look at that repo - but it seems like it doesn't work, and hasn't had much activity in a few years?

There was a PR from last year - but I don't see much after that?

https://github.com/embassy-rs/nrf70/pull/2

I'm definitely not qualified enough by far to help out on a technical basis here though unfortunately =(.

Not sure if donating hardware at all would help? I'm happy to pitch in to get some [nrF7002 DKs](https://www.nordicsemi.com/Products/Development-hardware/nRF7002-DK) to any team members, if that would help in any small way? Or some kind of bounty?

Wassasin, 2025-03-21:
As mentioned in https://github.com/embassy-rs/nrf70/pull/2 integrating the nRF700x is complex. With acceptable substitutes such as running ESP32 is hosted mode there has not (yet) been a compelling reason for someone to invest the time into implementing support for the nRF700x.

If you have a compelling reason to have it implemented by some other party feel free to DM me. There are a few companies that would be willing to do the work if you hire them.

simengangstad, 2025-06-28:
Just chipping in here. I've done some work on a fork here: https://github.com/simengangstad/nrf70/tree/main.

It can do basic scans now at least, but for some reason I'm getting the data out in a different format than what the wifi shell sample in Zephyr gets. The fork handles events and the various UMAC commands. 

I'm not too sure how much more effort I'll put into this without more documentation from Nordic. I've mostly worked from the Zephyr driver and comparing SPI transactions between the wifi shell example in Zephyr and this driver, but it's going very slow. 

There is a lot more functionality left to implement.