embassy-rs/embassy #3960: Cyw43 read address sometimes misaligned

Abestanis, 2025-03-11:
I was experiencing random crashes after 24+ hours of running a project, and the panic points to an assert in the `cyw43` crate:
```
cyw43-0.3.0\src\bus.rs:163:9:
assertion failed: addr % 4 == 0
```
I've since then disabled the `firmware-logs` feature and not experienced that panic since.

My guess is that the initial `bp_read` in `log_read` reads some garbage which is why the second one gets a misaligned address, but I'm not too sure.

In any case though I don't think a misaligned address there is an unrecoverable error and justifies panicking. Could we either not panic and log instead or not panic and just try to read the misaligned address?

Dirbaio, 2025-03-12:
I mean this divider https://github.com/embassy-rs/embassy/blob/38f26137fc67beb874aa73c9a7ab2150d9f3d372/examples/rp/src/bin/wifi_tcp_server.rs#L67

Abestanis, 2025-03-13:
Ah, I misunderstood, sorry. I'm also using `DEFAULT_CLOCK_DIVIDER`, so I still would assume it would work without problems with this default clock divider.

nui, 2025-05-06:
I got same issue on Raspberry pi pico 2w when using `DEFAULT_CLOCK_DIVIDER`.

Using `RM2_CLOCK_DIVIDER` work without problems.

i509VCB, 2025-06-25:
I appear to see this issue immediately after flashing with a Pico 2W. Using `RM2_CLOCK_DIVIDER` does fix it. Should we just set that divider by default so that the examples work (although this does slow stuff down)?

iwanders, 2025-07-12:
I'm just chiming in with another sample point, Pico 2W without any external components, flashed over USB with `picotool 2.1.1`, compiled with `rustc 1.88.0`.
To get a blinking LED I needed `RM2_CLOCK_DIVIDER`, as well as the `task-arena-size-40960` executor feature that was mentioned [here](https://github.com/embassy-rs/embassy/issues/2274#issuecomment-1868147748). Without that feature it stopped just before [this line](https://github.com/embassy-rs/embassy/blob/9651cfca51a273ba46d34ce8197fc0e63389b09e/examples/rp235x/src/bin/blinky_wifi.rs#L72) (at least, as identified with printing to a usb serial port).

Switching the divider sounds reasonable, with a comment that it may be possible to increase it but the example uses this for better compatibility?
Perhaps that arena size feature should also be enabled such that works out of the box? Docs for that feature imply it may not be necessary on [nightly](https://docs.rs/embassy-executor/latest/embassy_executor/#task-arena), perhaps that's why some people need that and others don't?