embassy-rs/embassy #3962: STM32 RTC Low Power stop_wakeup_alarm clears all set EXTI pending sources, not just it's own

garrettnmary, 2025-03-12:
The RTC stop_wake_alarm function performs a read/modify/write operation on the PR1 register, but the register is defined as a rc_w1, meaning that bits will clear when a 1 is written to them.  As such, the following line will clear all pending exti sources, not just the RTC one:
                crate::pac::EXTI
                    .pr(0)
                    .modify(|w| w.set_line(RTC::EXTI_WAKEUP_LINE, true));

This causes the EXTI input IRQ handler to not see the associated EXTI bit set in the PR1 register (embassy-stm32\src\exti.rs on_irq function), since it calls through this code-path before checking the pr(0) bits.  
Once changing the line in stop_wake_alarm to a write operation from a modify, in order to only write 1 to the RTC PR source, I get expected results.

STM32 features in use:
embassy-stm32 = { version = "0.1.0", path = "../embassy/embassy-stm32", features = [ "chrono", "exti", "low-power", "memory-x", "stm32l431vc", "time-driver-any", "unstable-pac", "low-power-debug-with-sleep"]  }

I've attached the PR1 register definition.

[image]

brightbloom, 2025-04-02:
This definitely seems worth fixing! And it's probably the true root cause of https://github.com/embassy-rs/embassy/pull/3404, which has been fixed for a while, but I'd never understood _how_ the IRQ pending bits were ending up cleared.

garrettnmary, 2025-04-03:
Yup, definitely seems like the same bug.  The benefit of fixing the above is other EXTI source (non I/O or RTC) that called through stop_wake_alarm can benefit as well.  Somehow I got myself on a build that didn't have the #3404 fix.  Will look into updating when our project has the chance.