embassy-rs/embassy #3967: Timer async delays lock up under certain circumstances

chmanie, 2025-03-14:
Hey,

first of all I'm _really_ sorry for the code to reproduce this issue. I tried everything I could but eventually it lead me to opening an issue here, so please bear with me here.

The issue is that (as far as I can tell) the `Timer` delays (`Timer::after_ms()`, etc.) lock up under _very_ certain circumstances (and I have only witnessed this on the rp2350). In the example this results in the logs just ceasing to happen.

It's really not deterministic, but it seems like it's happening when there's a lot of backpressure and/or lots of Futures trying to access the `Timer`.

I created a repository to reproduce the issue here: https://github.com/chmanie/lockup-16/. It can run on an rp2350 using `cargo run --release`.

Again, please excuse the code, I really tried to simplify as much as I could while having it reliably lock up in under a minute. The example is _very_ contrived but please be assured that this happened to me in less "imaginative" scenarios. I tried to apply all kinds of alterations, like only using one core, but ultimately it always freezes, sometimes sooner, sometimes later.

The code here spawns an "App" task 16 times in an executor on core 1, which has 16 slots. The apps all get a channel that lives on core 1 (`NoopRawMutex`) and feeds into another channel which then forwards them to core 0. I established this funnel architecture after having problems with too many CS mutexes being accessed by many tasks at the same time (but ultimately, maybe it was just the same problem). Then, all the tasks just hammer messages into the funnel every 10ms.

Should this be expected behavior, is there anything I can do to prevent this from happening? Are there patterns that I should avoid? How much is ultimately too much here?

chmanie, 2025-03-15:
I ran the same code (more or less, adapted to platform) on an rp2040 and on an esp32 with no issues. There are `rp2040` and `esp32s3` branches in the repo linked above.

Dirbaio, 2025-03-16:
this is most likely https://github.com/embassy-rs/embassy/commit/9da04cc38ea5cc17740bd9921f9f5cbb1c689a31 , fixed in embassy-rp v0.3.1. Your repro is using v0.3.0.

chmanie, 2025-03-16:
It certainly looks like it is. Thank you so much, this was driving me insane.

closed 2025-03-16 by chmanie