embassy-rs/embassy #3974: STM32H750VB MDMA peripheral support?

iamadahn, 2025-03-19:
I don't know if this is a bug or this peripheral is not supported, but embassy-stm32 crate for stm32h750vb chip doesn't have MDMA in it's peripherals, although it is in fact present and available for configuration in cubemx for example.

Dirbaio, 2025-03-19:
it's not a bug, it's just not implemented. PRs would be welcome.

iamadahn, 2025-03-20:
I'm afraid it's too much for me since I'm only learning how to Rust, but I can make a PR with example of how to use qspi with memory map for stm32h750vb chip. I had troubles with existing stm32h7b0 example since it uses octospi, and example for stm32l432kc with qspi didn't work for me at all.

ost-ing, 2025-05-21:
Its been added to the [stm32h7xx_hal](https://github.com/stm32-rs/stm32h7xx-hal/) repo, could probably be adapted from that. I could give it a shot soon

iamadahn, 2025-05-22:
It would be very nice, I could then test it with my board, which is WeAct stm32h750vb, and provide feedback if needed.