embassy-rs/embassy #3984: HSI48 for stm32l471rg

lelongg, 2025-03-20:
Previous versions of embassy allowed to set `config.rcc.clk48_src = Clk48Src::PLL1_Q;` but it doesn't exist anymore, and what seems to be the new syntax is not available for stm32l471rg :

```
config.rcc.mux.clk48sel = mux::Clk48sel::HSI48;
```

Dirbaio, 2025-03-20:
> config.rcc.mux.clk48sel = mux::Clk48sel::HSI48;

this is not available because stm32l471rg has no hsi48.

What do you need clk48 for? that L4 doesn't have USB.

lelongg, 2025-03-21:
I was porting code from an old embassy version, but I didn't understand why this particular part was used. The original issue was that `Adc::new` was blocking indefinitely. I traced the problem to the clock configuration and initially assumed it was due to this missing configuration. However, the issue was resolved by adding `config.rcc.mux.adcsel = Adcsel::SYS;`.

I'm still experiencing an issue with RNG blocking indefinitely, but I haven't identified the cause yet.

closed 2025-03-21 by lelongg