embassy-rs/embassy #3986: Old `embedded-hal` versions should be optional

inthehack, 2025-03-21:
# Problem

Multiple crates from Embassy depend both on `embedded-hal@1.0.x` and `embedded-hal@0.2.x`.

This generates a lot of useless and false positives while using `cargo deny` for instance. This also increases the build time and, less often, the binary size in some situations.

# Proposal

Because `embedded-hal@0.2.x` will be deprecated somtimes, I propose to:
- replace every `embedded-hal-1` dependency with bare `embedded-hal` version 1.0.0
- make `embedded-hal-02` optional in every crate in Embassy that depends on it

# Scope

The following crates will be impacted:
- `embassy-embedded-hal`
- `embassy-executor`
- `embassy-net-adin1110`
- `embassy-nrf`
- `embassy-rp`
- `embassy-stm32`

And:
- `examples/*`

lulf, 2025-03-21:
Binary size should only be affected if you actually use the embedded-hal 0.2 traits, do you have an example to the contrary?

There's still plenty of drivers using e-h 0.2, so I think it's premature to make it optional just yet.

inthehack, 2025-03-22:
> what's the problem with cargo-deny? why is it warning about eh0.2?

Hi, thanks for the reply.

The problem with `cargo-deny` appears as soon as one put deny on `multiple-versions`, which actually is a good code hygiene from many perspectives.

For instance, `embassy-stm32` implements traits for both versions if I am not wrong. Then, this crate adds both 0.2.x and 1.0.x verisons of `embedde-hal`. This makes `cargo-deny` fire an alarm with the configuration above.

Dirbaio, 2025-03-24:
> as soon as one puts a multiple-versions deny rule

Okay so it's warning because you've configured it to do so. Having two versions of the same crate is not necessarily bad in and of itself. It's bad in some cases, but not this case. You should configure an exception here.

The problem with gating eh0.2 behind a Cargo feature is it hurts ease of use significantly. Traits are a relatively advanced part of Rust, it's not obvious from looking at rustdocs that a driver is on another e-h version, and there's no way to make the compiler errors direct the user towards enabling the eh0.2 Cargo feature. 

Embassy's goal is to make things Just Work as much as possible. Having the two versions on by default makes all drivers Just Work regardless of which e-h version they use.

closed 2025-03-24 by Dirbaio