embassy-rs/embassy #3988: embassy_net: Add neighbors to cache manually.

Frostie314159, 2025-03-21:
Hi,
I was wondering if it would be possible for a net device implementation to add neighbors to the cache manually. The device I'm implementing is for Apple Wireless Direct Link (AWDL), which does not use NDP, but instead derives the IPv6 addresses from the MAC address of peers discovered at the Link Layer. As a workaround, I'm going to inject ICMPv6 Neighbor adverts into the stack, but it would be nice to have methods on the `StateRunner`, to add these manually.
Greetings
Simon

Dirbaio, 2025-03-21:
is this standard SLAAC or is it an Apple custom proprietary garbage thing? If it's SLAAC there's a WIP PR in smoltcp https://github.com/smoltcp-rs/smoltcp/pull/1039

Even if the IP addr is derived from the MAC, you should be able to use NDISC to discover them so you shouldn't need to manually fill the neighbor table.

Frostie314159, 2025-03-21:
I'm afraid it's the latter. AWDL is based on the IEEE 802.11 and is sort of like P2P Wi-Fi. Every device announces its presence through 802.11 action frames, and all other devices derive the IPv6 address solely from the peer's MAC address. When smoltcp sent neighbor solicitations, no response was ever transmitted. I am thinking about intercepting solicitations in the TX task and inject responses.

AWDL is sadly a proprietary protocol, so all information about it is based on RE by the TU  Darmstadt.

Dirbaio, 2026-08-29:
fixed by #6860 . there's APIs to manipulate the neighbor cache directly now.

closed 2026-08-29 by Dirbaio