embassy-rs/embassy #3995: STM32 C0 ADC always uses PA0

chudsaviet, 2025-03-23:
@karimpanacci reported:

@chudsaviet hi, I would love to do that but I don't really know where to start... I can understand the reference manual but I don't really know how to write code for a library like embassy, ​​I'm pretty new to Rust.

However I noticed another problem, which is that it seems that regardless of the ADC pin you select, it always uses PA0...

let mut adc = Adc::new(p.ADC1, SampleTime::CYCLES12_5, Resolution::BITS12);
let mut pin = p.PA1.degrade_adc();

loop {
    let blocing_pin = adc.blocking_read(&mut pin);
    info!("pa1 = {}.", blocing_pin);
}

Reading occurs but it is on the PA0 pin even if I set PA1

chudsaviet, 2025-03-23:
Hmm, I first thought pin number definition in `stm32-data` could be incorrect, but it looks fine - https://github.com/embassy-rs/stm32-data/blob/4a964af03b298de30ff9f84fcfa890bcab4ce609/data/registers/syscfg_c0.yaml#L432-L437.

rthinman, 2025-04-23:
Here's a little more information.  I'm starting a project with the STM32C031C6 Discovery Kit, which has a 5-position Dpad-switch with resistive dividers connected to PA4.  For background, I'm using RTIC and embassy-stm32 as the HAL.  I adapted the Embassy C0 adc.rs example to RTIC and tried the three read methods:

* `blocking_read()` reads PA0 when I pass it PA4.  To confirm it is reading from PA0, I added a jumper between the two pins to put an analog signal on the PA0 pin.  It returns the expected values when I manipulate the switch.
* `read()` using a one-element pin sequence always reads 0 for me, though I may be doing something else wrong.
* `read_in_hw_order()` works as expected: I give it a `u32` with just PA4's bit set, and I get a believable value returned in the buffer.  

`read_in_hw_order()` does what I need, so I'm not planning to investigate further at the moment.