embassy-rs/embassy #3997: Peripheral trait unsound: Mutexes can clone peripherals in safe code.

agrif, 2025-03-24:
I'm pretty sure I found a soundness issue with the Peripheral trait. Mutex guards implement DerefMut, and so guards that contain a peripheral implement the Peripheral trait. You can then get a PeripheralRef out of the guard, which drops the lock on the mutex, allowing you to borrow it again immediately.

`````rust
static PERI_MUTEX: Mutex<Option<PERI>> = Mutex::new(None);

fn main() {
    // initialize our peripheral mutex, e.g. via hal::init()
    {
        *PERI_MUTEX.lock().unwrap() = Some(PERI);
    }

    // MutexGuards implement DerefMut, and therefore Peripheral
    // but turning it into a PeripheralRef drops the guard...
    let ref1: PeripheralRef<'static, PERI> =
        { MutexGuard::map(PERI_MUTEX.lock().unwrap(), |o| o.as_mut().unwrap()).into_ref() };

    // so we can just borrow it again immediately
    let ref2: PeripheralRef<'static, PERI> =
        { MutexGuard::map(PERI_MUTEX.lock().unwrap(), |o| o.as_mut().unwrap()).into_ref() };

    println!("ref1 {:?} ref2 {:?}", *ref1, *ref2);
}
`````

[(playground with supporting code from embassy_hal_internal::peripheral)](https://play.rust-lang.org/?version=nightly&mode=debug&edition=2024&gist=2c9e444a1064133e391db8e7648eb27e)

My understanding is that holding two active `PeripheralRef<'static, PERI>` tokens for the same peripheral is a no-no, and breaks the assumption that only one thing has mutable access to a given peripheral at a time.

The most obvious fix would be to restrict the Peripheral trait implementation specifically to `&mut P` rather than `DerefMut<Target = P>`, but I don't know how much damage that would cause or what other legitimate uses for the DerefMut implementation there are.

Morgane55440, 2025-03-24:
i would like to add that not only does `MutexGuard` break here, but basically any type which has any Drop glue is likely to break this.

for example one might want to make a wrapper that holds a peripheral, and perform cleanup on drop. this would cause the cleanup to happen whenever `into_ref` is called

using [`core::mem::forget`](https://doc.rust-lang.org/nightly/core/mem/fn.forget.html) has a decent chance to fix the issue, but seems likely make things quite complicated for end users.

i would recommend just making it for `&mut P`

Dirbaio, 2025-03-27:
fixedin #3999

closed 2025-03-27 by Dirbaio