embassy-rs/embassy #4003: Flash operations in task makes RTT silently fail

stargazing-dino, 2025-03-25:
When using defmt with RTT for logging in the RP2040 flash example, moving flash operations from main to a spawned task causes RTT logging to silently fail

Reproduction

1. Running the original flash.rs example works correctly with RTT logging visible
2. When modifying the code to move all flash operations to a spawned task, only logs from main function appear

Actual:

```sh
cargo run --bin flash
    Compiling embassy-rp-examples v0.1.0 (/Users/katzen/Documents/embassy/examples/rp)
    Finished `dev` profile [optimized + debuginfo] target(s) in 1.66s
      Running `probe-rs run --chip RP2040 target/thumbv6m-none-eabi/debug/flash`
      Erasing ✔ 100% [####################]  20.00 KiB @  68.34 KiB/s (took 0s)
  Programming ✔ 100% [####################]  20.00 KiB @  38.69 KiB/s (took 1s)
              Finished in 0.81s
0.000366 [INFO ] Hello World! (flash src/bin/flash.rs:41)
```

Only the "Hello World!" message from main appears. None of the logs from the spawned task are visible

Expected:

```sh
➜ cargo run --bin flash
   Compiling embassy-rp-examples v0.1.0 (/Users/katzen/Documents/embassy/examples/rp)
    Finished `dev` profile [optimized + debuginfo] target(s) in 1.66s
     Running `probe-rs run --chip RP2040 target/thumbv6m-none-eabi/debug/flash`
      Erasing ✔ 100% [####################]  20.00 KiB @  70.21 KiB/s (took 0s)
  Programming ✔ 100% [####################]  20.00 KiB @  36.60 KiB/s (took 1s)                                                                  Finished in 0.83s
0.000365 [INFO ] Hello World! (flash src/bin/flash.rs:19)
0.010632 [INFO ] jedec id: 0xef4015 (flash src/bin/flash.rs:31)
0.010821 [INFO ] unique id: [230, 97, 48, 16, 15, 90, 33, 38] (flash src/bin/flash.rs:36)
0.010879 [INFO ] >>>> [erase_write_sector] (flash src/bin/flash.rs:76)
0.011438 [INFO ] Addr of flash block is 10100000 (flash src/bin/flash.rs:80)
0.011476 [INFO ] Contents start with [218, 218, 218, 218] (flash src/bin/flash.rs:81)
0.044856 [INFO ] Contents after erase starts with [255, 255, 255, 255] (flash src/bin/flash.rs:86)
0.053590 [INFO ] Contents after write starts with [218, 218, 218, 218] (flash src/bin/flash.rs:98)
0.053942 [INFO ] >>>> [multiwrite_bytes] (flash src/bin/flash.rs:48)
0.054472 [INFO ] Addr of flash block is 10101000 (flash src/bin/flash.rs:52)
0.054507 [INFO ] Contents start with [1, 2, 3, 4] (flash src/bin/flash.rs:53)
0.084500 [INFO ] Contents after erase starts with [255, 255, 255, 255] (flash src/bin/flash.rs:58)
0.087114 [INFO ] Contents after write starts with [1, 2, 3, 4] (flash src/bin/flash.rs:69)
0.087173 [INFO ] >>>> [background_read] (flash src/bin/flash.rs:105)
0.087316 [INFO ] Addr of flash block is 10102000 (flash src/bin/flash.rs:110)
0.087352 [INFO ] Contents start with daba1234 (flash src/bin/flash.rs:111)
0.117122 [INFO ] Contents after erase starts with ffffffff (flash src/bin/flash.rs:116)
0.117794 [INFO ] Contents after write starts with daba1234 (flash src/bin/flash.rs:130)
```

<details>
  <summary>Diff</summary>

```diff
diff --git a/examples/rp/src/bin/flash.rs b/examples/rp/src/bin/flash.rs
index eb3e6a2b9..456419e91 100644
--- a/examples/rp/src/bin/flash.rs
+++ b/examples/rp/src/bin/flash.rs
@@ -13,18 +13,9 @@ use {defmt_rtt as _, panic_probe as _};
 const ADDR_OFFSET: u32 = 0x100000;
 const FLASH_SIZE: usize = 2 * 1024 * 1024;
 
-#[embassy_executor::main]
-async fn main(_spawner: Spawner) {
-    let p = embassy_rp::init(Default::default());
-    info!("Hello World!");
-
-    // add some delay to give an attached debug probe time to parse the
-    // defmt RTT header. Reading that header might touch flash memory, which
[... 162 lines cut ...]
    info!("Contents after erase starts with {=u32:x}", buf[0]);
    if buf.iter().any(|x| *x != 0xFFFFFFFF) {
        defmt::panic!("unexpected");
    }

    for b in buf.iter_mut() {
        *b = 0xDABA1234;
    }

    defmt::unwrap!(flash.blocking_write(ADDR_OFFSET + offset, unsafe {
        core::slice::from_raw_parts(buf.as_ptr() as *const u8, buf.len() * 4)
    }));

    defmt::unwrap!(flash.background_read(ADDR_OFFSET + offset, &mut buf)).await;
    info!("Contents after write starts with {=u32:x}", buf[0]);
    if buf.iter().any(|x| *x != 0xDABA1234) {
        defmt::panic!("unexpected");
    }
}
```
</details>

Details

I'm using a pico debug probe with standard 2040 pico w

lulf, 2025-03-25:
Your `main()` contains a `loop{}` at the end. This will block other tasks from running on the executor. Just remove the loop{} and you should be fine. The flash task will still run.

stargazing-dino, 2025-03-25:
I tried everything but the obvious xD

Thank you for your help. Yes it was definitely that.

closed 2025-03-25 by stargazing-dino