embassy-rs/embassy #4006: Can't compile because of atomic-polyfill and embedded-storage-async require -Z unstable-options

FransUrbo, 2025-03-25:
Not entirely sure what and where the problem is, but the only place left in Embassy that still uses `atomic-polyfill` is `embassy-rp`, all the other have removed it years (?) ago..

```
$ cargo build
   Compiling pio-parser v0.3.0
   Compiling atomic-polyfill v1.0.3
   Compiling embedded-storage-async v0.4.1
   Compiling embassy-executor v0.7.0
   Compiling num-traits v0.2.18
   Compiling fixed-macro-impl v1.2.0
   Compiling smart-leds-trait v0.3.1
   Compiling embassy-futures v0.1.1
error: the `-Z unstable-options` flag must also be passed to enable the flag `check-cfg`

error: could not compile `embedded-storage-async` (lib)

Caused by:
  process didn't exit successfully: `rustc --crate-name embedded_storage_async --edition=2018 <HOME_DIR>/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embedded-storage-async-0.4.1/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no -C codegen-units=1 -C debuginfo=2 -C debug-assertions=on --check-cfg 'cfg(docsrs,test)' --check-cfg 'cfg(feature, values())' -C metadata=26db367fa9c52522 -C extra-filename=-23a9610a163ebed8 --out-dir '<SOURCE_DIR>/target/thumbv6m-none-eabi/debug/deps' --target thumbv6m-none-eabi -L 'dependency=<SOURCE_DIR>/target/thumbv6m-none-eabi/debug/deps' -L 'dependency=<SOURCE_DIR>/target/debug/deps' --extern 'embedded_storage=<SOURCE_DIR>/target/thumbv6m-none-eabi/debug/deps/libembedded_storage-24a0a19d36cd7344.rmeta' --cap-lints allow -C linker=flip-link -C link-arg=--nmagic -C link-arg=-Tlink.x -C link-arg=-Tlink-rp.x -C link-arg=-Tdefmt.x -C no-vectorize-loops` (exit status: 1)
warning: build failed, waiting for other jobs to finish...
error: the `-Z unstable-options` flag must also be passed to enable the flag `check-cfg`

error: could not compile `atomic-polyfill` (build script)

Caused by:
  process didn't exit successfully: `rustc --crate-name build_script_build --edition=2021 <HOME_DIR>/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/atomic-polyfill-1.0.3/build.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C codegen-units=8 -C debug-assertions=off --check-cfg 'cfg(docsrs,test)' --check-cfg 'cfg(feature, values())' -C metadata=bac5af936ed4a0ad -C extra-filename=-ace0ff1cf0fc3691 --out-dir '<SOURCE_DIR>/target/debug/build/atomic-polyfill-ace0ff1cf0fc3691' -C strip=debuginfo -L 'dependency=<SOURCE_DIR>/target/debug/deps' --cap-lints allow` (exit status: 1)
$
```

But `embedded-storage-async` is still used everywhere.. I *REALLY* (!!) don't want to have to use `-Z unstable-options`, because I don't want to use a nightly! Also, I get millions of other problem when I do.. :D

So I'd very much appreciate some suggestions..

My `Cargo.toml` (relevant sections). The `portable-atomic` dependency is supposed to replace `atomic-polyfill`, but it doesn't seem to *provide* `atomic-polyfill`..

```
[dependencies]
critical-section = "1.2.0"
portable-atomic = { version = "1.10.0", features = ["critical-section"] }

[dependencies.embassy-embedded-hal]
version = "0.3.0"
features = ["defmt"]

[dependencies.embassy-executor]
version = "0.7.0"
features = ["task-arena-size-32768", "arch-cortex-m", "executor-thread", "executor-interrupt", "defmt"] #, "integrated-timers"]

[dependencies.embassy-time-driver]
version = "0.2.0"

[dependencies.embassy-time]
version = "0.4.0"
features = ["defmt", "defmt-timestamp-uptime"]

[dependencies.embassy-rp]
version = "0.4.0"
features = ["defmt", "unstable-pac", "time-driver", "critical-section-impl"]

[dependencies.embassy-sync]
version = "0.6.2"
features = ["defmt"]
```

Dirbaio, 2025-03-25:
Try updating Rust.

FransUrbo, 2025-03-25:
Thanx, but as far as I can tell, I'm running the latest version..
```
$ <HOME_DIR>/.cargo/bin/rustc --version
rustc 1.85.1 (4eb161250 2025-03-15)
```

Dirbaio, 2025-03-25:
You have multiple rustc's installed then, the one `cargo build` is picking up is old. Try `rustc --version`, try `rustup update`.

FransUrbo, 2025-03-25:
Done that..
```
$ which -a rustc
<HOME_DIR>/.cargo/bin/rustc
/usr/local/bin/rustc
$ <HOME_DIR>/.cargo/bin/rustc --version
rustc 1.85.1 (4eb161250 2025-03-15)
$ /usr/local/bin/rustc --version
rustc 1.85.1 (4eb161250 2025-03-15) (Homebrew)
$
```

FransUrbo, 2025-03-26:
Managed to get it compiling by using `main#9edf5b7`..

lulf, 2025-03-31:
Glad you got it working!

closed 2025-03-31 by lulf