embassy-rs/embassy #4008: How to build without defmt?

akavel, 2025-03-26:
I need to debug a hairy low-level issue. I'm looking through assembly in objdump, but huge amounts of the code seems busy with various defmt stuff I'm not using at all, and it makes it hard for me to see anything relevant to my problem. Is it possible to somehow build without defmt at all? Specifically, I'm interested in code built with `embassy-nrf`.

lulf, 2025-03-26:
As long as you don't enable the 'defmt' feature, and/or don't link to defmt in linker scripts/build.rs, you won't 'build with defmt'. Could you share more of your setup maybe? Cargo.toml and build.rs for instance.

akavel, 2025-03-26:
Ahhh, thanks a lot! It didn't come to my mind to look also in features. After disabling it, I still had some "core::fmt" show up in the objdump, but then I found out I didn't actually enable basic optimizations in this project, so after adding `opt-level = "z"` and `lto = true` the "blinky" dump is down from 3k lines to 1k lines, and no longer any `core::fmt` distractions there. Still a chunky bite, but at least presumably only relevant stuff left. Thanks again!