embassy-rs/embassy #4018: Questions about STM32 adc driver layout.

Sandvoxel, 2025-03-28:
Hello Embassy Team,

I have a question regarding the structure of the STM32 ADC driver. I understand that the design must accommodate specifying the pin for measurement, but I have noticed some inconsistencies compared to other drivers.

Currently, there is a trait for a single channel, AdcChannel, and an abstraction, AnyAdcChannel. It appears that AnyAdcChannel has not been updated with the new PeripheralType trait, which prevents its use in contexts such as:

```rust
voltage_sense: Peri<'a, impl AdcChannel<T>>,
```
This oversight limits the ability to store ADC channels in structs. While I understand that the degrade_adc function allows conversion between types, I believe it should be possible to directly use voltage_sense.into(), similar to how other peripherals operate.

Thank you for your time and assistance.

I am currently working with the relatively new adc driver for the stm32u5.

Sandvoxel, 2025-03-28:
For some more context I'm trying to make a general purpose struct for handling power buses. I'm also willing to concede that I may just be going out this the wrong way. It just feels so wrong making it store the specific peripheral instead of the generalized definition.

i509VCB, 2025-03-29:
Looking at the docs, it appears that `AnyAdcChannel` implements PeripheralType: https://docs.embassy.dev/embassy-stm32/git/stm32u595ri/adc/struct.AnyAdcChannel.html#impl-PeripheralType-for-AnyAdcChannel%3CT%3E

You might need an `into()` to go from the concrete adc channel type to an `AnyAdcChannel`

Sandvoxel, 2025-04-01:
Doing this as sugested leads to a different Error as Peri Expects a trait.

```rust
impl<'a, T: Instance> VoltageMonitor<'a, T> {
    pub fn new(
        enable: Output<'a>,
        reset: Output<'a>,
        voltage_sense: Peri<'a, impl AnyAdcChannel>,
        current_sense: Peri<'a, impl AnyAdcChannel>,
    ) -> Self {

        Self {
            enable,
            reset,
            voltage_sense: voltage_sense.into(),
            current_sense: current_sense.into(),
        }
    }
}
```

```
error[E0404]: expected trait, found struct `AnyAdcChannel`
   --> src/voltage_monitor/mod.rs:20:38
    |
20  |         voltage_sense: Peri<'a, impl AnyAdcChannel>,
    |                                      ^^^^^^^^^^^^^ help: a trait with a similar name exists: `AdcChannel`
    |
   ::: /Users/kylekovalchick/Documents/projects/embassy/embassy-stm32/src/adc/mod.rs:142:1
    |
142 | pub trait AdcChannel<T>: SealedAdcChannel<T> + Sized  {
    | ---------------------------------------------------- similarly named trait `AdcChannel` defined here

```

leftger, 2026-11-05:
Closing this as resolved by the ADC driver rework. [#6997](https://github.com/embassy-rs/embassy/pull/6997) (\"stm32/adc: single unified driver\") replaced `AnyAdcChannel` with `BorrowedAdcChannel`.

To store a channel without naming the pin type:

- Each pin is an `AdcChannel<'d, T>` as `Peri<'d, PIN>`.
- `degrade_adc()` consumes the pin and returns a `BorrowedAdcChannel<'d, T>`, which holds the channel number and a lifetime.
- `Adc::read` and `Adc::blocking_read` take `impl BorrowedChannel`, so a stored `BorrowedAdcChannel` can be passed directly.

The `Peri<'a, impl AdcChannel<T>>` plus `.into()` form from the original report is not part of this API. The erased type is not a `PeripheralType`, so it cannot be taken as a `Peri` argument. Storing the channel with `degrade_adc()` in a struct field covers the use case.

closed 2026-11-05 by leftger