embassy-rs/embassy #4030: executor: WFI in critical section never resumes (riscv)

outfoxxed, 2025-03-31:
I've been writing a time driver for the WCH CH583 chip, which uses WCH's QingKe V4A (riscv32imac) architecture, and the systick interrupts aren't working correctly under embassy, while they work fine in a standalone project. I've traced this to the WFI instruction inside a critical section linked below.

https://github.com/embassy-rs/embassy/blob/a44abaf7e4562fa5393087fd845bf0d02141325b/embassy-executor/src/arch/riscv32.rs#L64-L74

The critical section implementation in the qingke crate disables all interrupts, which means the WFI will never be interrupted.

https://github.com/ch32-rs/qingke/blob/73a21d1b527327c17bd80b8722d66e10a4fcd46e/src/critical_section_impl.rs#L7-L18

Excerpt from the processor manual:
[image]

Changing the embassy executor snippet above to WFI outside of the critical section results in interrupts working as expected.
```rs
let dowfi = critical_section::with(|_| {
    // if there is work to do, loop back to polling
    // TODO can we relax this?
    if SIGNAL_WORK_THREAD_MODE.load(Ordering::SeqCst) {
        SIGNAL_WORK_THREAD_MODE.store(false, Ordering::SeqCst);
        false
    }
    // if not, wait for interrupt
    else {
        true
    }
});

if dowfi {
    core::arch::asm!("wfi");
}
```

I'm opening an issue for this instead of a PR as I assume there's some reason the WFI is happening inside a critical section, but I don't know what that reason could be.

Dirbaio, 2025-03-31:
riscv spec says an interrupt while interrupts disabled must still cause WFI to exit (but it's not handled right away, it's handled after exiting the critical section).

Maybe this is a bug in the ch583?

outfoxxed, 2025-03-31:
`mstatus` behaves exactly the same as `gintenr` here, so its likely a bug in the chip.

I've tested it in depth and found that WFI *does* respond to interrupts, but only if they happen in a short (couple ms or less) window from WFI being called inside a critical section.

Is there a way to work around this?

ExplodingWaffle, 2025-03-31:
Have you tried the WFITOWFE and SEVONPEND bits in the PFIC? They should make the interrupt behaviour of these cores more well behaved (specifically SEVONPEND states that disabled interrupts should wake the core)

(n.b. while looking at these PFIC registers again, it occurs to me that it might be possible to use the ripoff-SEV feature to write a more ARM-style executor without the need for the atomic dirbaio mentioned. obviously would be WCH specific though, probably not worth doing unless it’s the only way)

outfoxxed, 2025-03-31:
WFITOWFE and SEVONPEND combined work as expected.

Dirbaio, 2025-03-31:
You could make your own WCH-specific executor and put it in the HAL. Do not enable any `arch-*` Cargo features, copy `riscv.rs` into the HAL and customize. The `raw` module is public to allow 

So nice that they copied WFE/SEV from ARM. It's a really nice feature, raw `wfi` is a bit troublesome (also on ARM) 😢. With this you can remove the atomic, just like on ARM. 

(or you could set WFITOWFE and SEVONPEND and just tell the user to just use the `embassy-executor/arch-riscv` feature. It'll be slightly slower because of the now-useless atomic)

outfoxxed, 2025-04-01:
Thanks. I'll port it to the HAL as that seems more correct.

closed 2025-04-01 by outfoxxed