embassy-rs/embassy #4038: stm32 low-power: executor never wakes up when await'ing LPUART tx or rx

brightbloom, 2025-04-02:
LPUART peripherals don't contribute to `crate::rcc::REFCOUNT_STOP2` or `crate::rcc::REFCOUNT_STOP1`. This makes some sense, because LPUART peripherals are able to stay running when the core is asleep.

However, it leads to unintended consequences. DMA interrupts aren't wakeup events, which means when the UART is used in Async mode, the core will go to sleep waiting for the LPUART transaction to complete and never wake up.

Even if DMA interrupts were wakeup sources, the DMA controller is on the AHB interface, and the AHB clock is disabled during STOP modes. So, regardless of the wakeup issue, any ongoing LPUART transactions will lose bytes if the core goes to sleep.

My first thought is adding to 1 to `crate::rcc::REFCOUNT_STOP1` and `crate::rcc::REFCOUNT_STOP2` at the start of any Async UART transaction, and subtracting 1 at the end. Does this sound like the right solution?

brightbloom, 2025-04-02:
small language correction - describing this as the "core going to sleep" could imply the chip's "Sleep" mode, which the low-power executor doesn't use. This affects STOP0, STOP1, and STOP2.

danielstuart14, 2025-04-03:
Maybe we should increase the stop 1/2 refcount on the DMA (and GPDMA) implementation side, as for example stm32u5 has a LPDMA peripheral that can be on during "sleep" and should be able to work with lpuart.

Also, another thing I thought: even if DMA doesn't work, I can see use cases where we would want to wake-up on any new data from the lpuart peripheral (it would be great if we did not loose the data, but just waking up is already an achievement). 

So maybe we would need an "wait_data" method that is async (using IRQs) but doesn't do a DMA transfer.
Of course there is also the option to have "read" and "write" methods that do not use DMA and rely only on interrupts. 

Does it make sense?

danielstuart14, 2025-04-14:
An update, I tried just incrementing STOP1_REFCOUNT on the async write and read functions, which made it not "sleep" during transfers, I even tried waiting for the transfer to complete before decrementing the refcount.  But later when getting back from stop 1 mode (when not making a LPUART transfer), the serial would output garbage until a reset, so there is probably some need to reconfigure the DMA/peripheral after getting back from the stop mode.

There is also some issues with this naive approach: 
- We currently can't differentiate the DMA channel we're using on the USART driver (LPDMA works with STOP2 and GPDMA works with STOP1 on stm32u5. No DMA channel works with STOP1 on stm32h5), so we would have to restrict to the stop mode supported by all DMAs for the selected stm model.  
- Every single peripheral that uses DMA and is compatible with a stop mode would need this manually implemented. A better solution probably would be rebasing the DMA implementation to have this derived from stm32-data, but currently the DMA looses its type as soon as the peripheral is instanciated (instead a channel number is stored in the peripheral struct and used to perform transactions).

danielstuart14, 2025-04-16:
@Dirbaio any idea on the best way to move forward?

xoviat, 2025-11-21:
@danielstuart14 Are you still interested in working on this?