embassy-rs/embassy #4041: Discussion: Optional executor perf counters

jamesmunns, 2025-04-03:
In #4033, and as part of https://github.com/jamesmunns/postcard-rpc-embassy-trace, I did quite a bit of work using the `trace` hookpoints to transmit a stream of events to be collected and processed by a host computer.

This allowed me to obtain a detailed event timeline, as well as aggregate metrics like:

```text
EXECUTOR:   IDLE:  95.55%, POLL:   3.63%, SCHD:   0.81%
 MAIN_TASK: IDLE:  99.91%, ACTV:   0.06%, WAIT:   0.04%, WAKE: 0014
WRK-000000: IDLE:  99.51%, ACTV:   0.40%, WAIT:   0.08%, WAKE: 0017
WRK-000001: IDLE:  99.09%, ACTV:   0.41%, WAIT:   0.50%, WAKE: 0017
WRK-000002: IDLE:  98.65%, ACTV:   0.41%, WAIT:   0.93%, WAKE: 0017
WRK-000003: IDLE:  98.22%, ACTV:   0.42%, WAIT:   1.36%, WAKE: 0017
WRK-000004: IDLE:  97.77%, ACTV:   0.42%, WAIT:   1.81%, WAKE: 0017
WRK-000005: IDLE:  97.31%, ACTV:   0.43%, WAIT:   2.26%, WAKE: 0017
WRK-000064: IDLE:  99.90%, ACTV:   0.05%, WAIT:   0.06%, WAKE: 0005
WRK-00006E: IDLE:  99.93%, ACTV:   0.05%, WAIT:   0.02%, WAKE: 0017
DRAIN_TASK: IDLE:  97.98%, ACTV:   0.75%, WAIT:   1.27%, WAKE: 0739
  USB_TASK: IDLE:  98.61%, ACTV:   0.39%, WAIT:   1.00%, WAKE: 0781
  LOG_TASK: IDLE:  99.97%, ACTV:   0.01%, WAIT:   0.02%, WAKE: 0006
```

Showing:

* The number of tasks
* For the Executor:
    * The percentage of ticks spent idle
    * The percentage of ticks spent polling
    * The percentage of ticks spent scheduling
* For each task:
    * The name of the task (this was an extension)
    * The percentage of ticks spent idle
    * The percentage of ticks spent polling
    * The percentage of ticks spent waiting
    * The number of wakes

The downside of this is that streaming all of these events is relatively expensive: we need to send task ids and timestamps for every event so they can be properly attributed.

However, if we ONLY want the aggregate metrics, this can be done DRASTICALLY cheaper locally by storing information in the Executor itself, as well as the `TaskHeader` of each task. We would need to store:

* for the executor (32 bytes):
    * a u64 for "total ticks idle"
    * a u64 for "total ticks active"
    * a u64 for "total ticks scheduling"
    * a u64 for "last state change tick"
* for each task (40ish bytes):
    * a u64 for "total ticks idle"
    * a u64 for "total ticks active"
    * a u64 for "total ticks waiting"
    * a u64 for "last state change tick"
    * a u8 for "current state"
    * (optional) a pointer to a static string of the name of the task
    * a u32 (maybe u64) for "total wakes"

We could then manipulate the counters at the same points where we currently emit `trace` events.

I don't have a solution of "how you get that data", some ideas are:

* have the executor log it periodically, based on some interval + time of last logging
    * It could do this for itself, and check each task on poll
    * this would be "automatic", but tasks would need to be polled for it to work
* have the tasks trigger this themselves, using an interface like `Deadline`
    * would require in-band changes, like `Perf::log().await`
    * Or could be added as an outer biased select in the `#[task]` macro, checking on every poll?
* It would be neat to be able to trigger this "on demand" and dump all info, but this would require
  some way of sharing this data safely, which I'm not sure how.

There are some potential downsides (remembering that this would be an opt-in feature):

* This scheme (as-is) probably wouldn't work with interrupt executors
* This scheme will mis-account time spent in interrupts in general
* This scheme increases size of tasks and executor
* This scheme will add some perf/latency overhead for book keeping, though not significantly for atomic targets, and a few extra critical sections for non-atomic targets.
* This scheme is limited to the resolution of the `embassy-time-driver`, many targets use 32.768kHz clocks, which are roughly 30.5us. For example on the RP2040 at 150MHz, that would be 4578 CPU cycles, potentially too coarse for measuring some task polls at all. Not all targets offer a higher resolution 1MHz clock.

I wanted to dump my notes here, in case it was useful to discuss or implement.

lulf, 2025-04-03:
I made a variant of this at Akiles, where I built the counters on top of the trace hooks, would that work for this? Basically the app would have a 

```
static TASK_TRACE: Mutex<CriticalSectionRawMutex, RefCell<TaskTrace<40>>> = Mutex::new(RefCell::new(TaskTrace::new()));
```

Where TaskTrace contains aggregated stats about the task that gets updated by the trace hooks. This way you don't really need to change the executor code. All the data is aggregated in the TaskTrace, and you can 'print()' it when desireable. In our case, it was on a panic that this information is printed to the logs.

I could put the implementation up somewhere if you're interested.

jamesmunns, 2025-04-03:
I think "don't have any built in perf counters, rely on external crates using existing `trace` hooks" is a reasonable alternative to solving this problem as well. I think you can do almost everything you need through the existing hooks.

Two upsides I can think of immediately doing this inside embassy imo:

1. We can always "right size" the storage space for tasks without guessing, by making them part of the TaskHeader
2. We can benefit from having the executor "drive" the process, since it knows which task it is working on (no need for task IDs), and avoids extra synchronization/locking, since it (and tasks) know when they have exclusive access to data

I don't feel *super* strongly it needs to be built-in, but I think either having it as a built in feature flag, or having a first party crate folks can drop in, would be a fairly useful tool.

bagbag, 2025-04-03:
If embassy would track the instants of each event and pass them to the hooks, I think all of that could be handled externally and I wouldn't need to store any task_ids in maps inside a Mutex accessed via unsafe.