embassy-rs/embassy #4079: [RP24235x] Uart Rx await blocks application if no data received.

markusadamek, 2025-04-12:
When awaiting the buffered RX read_exact when the receive buffer is empty, the call will stall the application.

My expectation was that awaiting read_exact would allow other tasks to process while waiting for data. 

Is this intended behavior?

I added an example here:
https://github.com/markusadamek/pico-example/blob/main/src/main.rs

In this example the main loop toggling the LED will eventually stop if no data is provided to the UART0 RX.

markusadamek, 2025-04-23:
I was able to pin down where the error occurs:

If nothing is connected to the RX pin, rx.read_exact will return with ErrorCode Break. In my case this will panic the application (because of unwrap). It also seems to only happen once during startup, subsequent calls to rx.read_exact will wait as intended. I can only guess that this is because on subsequent calls the RX interrupt is never raised and the error flag is not updated. 

While it is somewhat odd behavior, one can assume it is intended. 

I think it might still be a good idea to change the example to actually handle the error instead of using unwrap within the reader task of the rp235x/uart_buffered_split.rs example. 
If nothing else to encourage proper error handling.

Dirbaio, 2025-04-23:
Are you using a debug probe to get logs from the firmware? `.unwrap()` should cause an error message to get printed that makes it obvious what's wrong.

markusadamek, 2025-04-23:
No, I was only using the usb logger

Dirbaio, 2025-04-23:
then it's expected you don't get panic messages, it's not the example's fault. I strongly recommend you get a debug probe.

closed 2025-04-23 by markusadamek