embassy-rs/embassy #4081: Error when trying to use wifi_blinky example code

ofluffydev, 2025-04-13:
Error:

```rust
error[E0609]: no field `rasr` on type `MPU`
   --> /home/<username removed>/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embassy-rp-0.4.0/src/lib.rs:541:18
    |
541 |         core.MPU.rasr.write(
    |                  ^^^^ unknown field
    |
    = note: available fields are: `_type`, `ctrl`, `rnr`, `rbar`, `rlar` ... and 7 others
```

I tried digging around and failed to find a solution, only difference in this code and the example code is the dependencies do not use source paths like the examples folder did.

Cargo.toml and full output attached

[out.txt](https://github.com/user-attachments/files/19722465/out.txt)

[Cargo.toml.txt](https://github.com/user-attachments/files/19722466/Cargo.toml.txt)

1-rafael-1, 2025-04-13:
Seen this error myself, check if you use the feature for the rp chip for embassy-rp correctly. Either rp2040 or rp235xa/ä or rp235xb. At a guess you are on Pico 2 but set to rp2040 instead?

ofluffydev, 2025-04-14:
i did rp235xb since it was the most used in the examples, not sure why
theres an A and a B version, this needs to be documented somewhere.

maadamek, 2025-04-15:
I had the same issue recently - turns out I didn't include the .cargo folder/dependencies into my project root.

1-rafael-1, 2025-04-18:
@ofluffydev Did You get this to work? 

If not... ran into the very same error myself today, while building embassy-rp. Then remembered this issue. Not sure what You were doing exactly, but in my case I just unwisely tried to build embassy-rp without aligning target and features. 

Looking at the cargo.toml You provided i looks like you are trying to set up a distinct own project?

Assuming that, I believe You are not compiling for the correct target. Can You show the contents of Your .vscode/config.toml?

U007D, 2025-08-19:
@ofluffydev there's an `rp235xa` and `rp235xb` because the "RP2350" comes in four flavors.

The 'B' variants have more pins on the package than the 'A' variants, providing more GPIO and more analalog ADC inputs.

For completeness, the 'x' in the feature names is because the "RP2350" is also available as an RP2354 (also with A and B variants).  The 2354 includes 2MB of on-die flash memory (the 2350 has none).

From the [RP2350 datasheet](https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf):
> Product Package Internal Flash GPIO Analogue Inputs
> RP2350A QFN-60 None 30 4
> RP2350B QFN-80 None 48 8
> RP2354A QFN-60 2 MB 30 4
> RP2354B QFN-80 2 MB 48 8

The Raspberry Pi Pico 2 (W) uses the RP2350A (use --features rp2350xa).

When building `embassy-rp` it does appear to select the wrong target ISA.  including  `--target thumbv8m.main-none-eabihf` fixed the "error[E0609]: no field `rlar` on type `MPU`" for me.

tl;dr: Try `cargo check --features rp235xa --target thumbv8m.main-none-eabihf`.