embassy-rs/embassy #4099: Extending the executor to allow integration with other runtimes

ThoFrankACN, 2025-04-15:
Hi there,

I am currently looking into adding Rust support into [OpenBSW](https://github.com/eclipse-openbsw/openbsw). I created a proof of concept using a `raw::Executor` which runs alongside other C++ Runnables inside one FreeRtos task using [OpenBSW's cooperative Async](https://eclipse-openbsw.github.io/openbsw/sphinx_docs/libs/bsw/async/doc/user/index.html).

The `raw::Executor` doesn't play too nicely with the other C++ runnables, because it only returns execution after all tasks are pending. Therefore I would like to extend / fork the executor to be able to potentially return after every await point.

Would this be something you would be willing to accept here?

On a different but related note: For automotive, safety is quite an important topic. There is the ISO 26262 standard for automotive and similar ones for other industries. Did you have any contact points with that already?

Lastly, thank you very much for your work. It really is a pleasure working with embassy!

Dirbaio, 2025-04-15:
> it only returns execution after all tasks are pending. Therefore I would like to extend / fork the executor to be able to potentially return after every await point.

Why do you need that? 

When `raw::Executor::poll()` polls a task, it calls the `poll()` method on the future, which runs until the future becomes pending. This can be one or many await points, if many are instantly ready. An await point doesn't necessarily yield. So there's no way to make it "return after every await point"

ThoFrankACN, 2025-04-15:
Because OpenBSW's Async is cooperative, it's necessary to return execution at least sometimes. If the executor doesn't return the other Runnables starve.

I wasn't aware that an await point doesn't always yield. But that should be fine. One could force regular yields with `yield_now`. And as soon as there is a task switch, it's possible to return to Async.

Dirbaio, 2025-04-15:
> If the executor doesn't return the other Runnables starve.

`raw::Executor::poll()` does return, when it has polled all tasks that need polling.

If a task never yields it's a bug that should be fixed. It'll starve all other async tasks, it's not just a problem for non-rust-async tasks.

ThoFrankACN, 2025-04-17:
True. I did some homework and pondering about all this.

Starving should not be an issue. Thanks for helping me understand that. The `raw::Executor::poll()` can only poll each task once at most. So without any bugs this eventually returns. Then other C++ Runnables can take over.

What remains is the fairness issue. It would be nice to be able to only pop the first task from the queue and then potentially execute a C++ Runnable afterwards. But I guess that would not be possible with the current RunQueue. Which means I would then need a different executor...

closed 2025-04-17 by ThoFrankACN