embassy-rs/embassy #4103: ADC VBAT channel on H5 and U0?

esden, 2025-04-17:
I was looking at some of the special ADC channels in STM32H5 and came across this code:

https://github.com/embassy-rs/embassy/blob/659c616f9f13aa6f157eaef5279f09805449b5be/embassy-stm32/src/adc/v3.rs#L60-L63

As far as I can tell the host checks should probably not be twice for the `adc_h5`. Probably one of them should be for `adc_u0` considering that the other channel implementations check for that group of devices.

Still at the end the VBat channel for the H5 does not seem to be correct. According to the reference manual: https://www.st.com/resource/en/reference_manual/rm0481-stm32h52333xx-stm32h56263xx-and-stm32h573xx-armbased-32bit-mcus-stmicroelectronics.pdf page 1039 the VBat channel is 16 not 2 or 13.

If you agree that there is a problem and I am not somehow confused I will be happy to put together a PR to fix this.

Dirbaio, 2025-04-17:
yeah seems like a mistake. PRs welcome as always :)

esden, 2025-04-17:
Ohh right, on top of it all we should think about a way to encode that a channel is available on a specific ADC or not. On the F4 parts it seems that only ADC1 has the special channels, while other parts have different special channels on ADC1 than ADC2... 

I don't think this has to be addressed immediately but it would be nice to encode what channel selections are valid in the type system, resulting in better UX. :)