embassy-rs/embassy #4122: STM32 ADC read with DMA doesn't preserve channel order

Mortano, 2025-04-23:
I am trying to configure an ADC with DMA for reading samples from some sensors on a custom PCB using an STM32F4-family chip. I followed the [ADC DMA example](https://github.com/embassy-rs/embassy/blob/main/examples/stm32f4/src/bin/adc_dma.rs) and configured the ADC sample sequence / channels like so:
```rust
adc.set_sample_sequence(
    Sequence::One,
    &mut peripherals.analog_pin_1,
    SampleTime::CYCLES112,
); // Sensor 1 at pin_1
adc.set_sample_sequence(
    Sequence::Two,
    &mut peripherals.analog_pin_2,
    SampleTime::CYCLES112,
); // Sensor 2 at pin_2
```

I then call `adc.read()` in a loop from a high-priority interrupt executor. I would expect that the buffer I pass to `adc.read()` contains samples in the order `[PIN1, PIN2, PIN1, PIN2, ...]`, but instead the order keeps changing every time I call `adc.start()` (e.g. by restarting my CPU). Sometimes it is the expected order, sometimes it is reversed (`[PIN2, PIN1, PIN2, PIN1, ...]`). 
The documentation however states that the order should be preserved:

"Each call to read will populate the measurements array in the same order as the channels defined with set_sample_sequence. There will be many sequences worth of measurements in this array because it only returns if at least half of the DMA buffer is filled. For example if 3 channels are sampled measurements contain: [sq0 sq1 sq3 sq0 sq1 sq3 sq0 sq1 sq3 sq0 sq1 sq3..]." (from `RingBufferedAdc::read`)

I am not seeing any overrun errors, `adc.read()` returns successfully. Did I misunderstand how to use the ADC with DMA?

tommasoclini, 2025-06-09:
I'm having this exact problem too, on stm32f722re

tommasoclini, 2025-06-12:
@Mortano what did you end up using for the ADC?

Mortano, 2025-06-18:
@tommasoclini I used the blocking ADC implementation as a workaround because I don't have a lot of data to read. Not ideal but acceptable for my usecase.

tommasoclini, 2025-06-25:
@Dirbaio do you have any idea of what the problem could be?