embassy-rs/embassy #4144: Is `nrf::radio::ble::Radio` unsound?

workingjubilee, 2025-04-28:
Let us consider [`embassy_nrf::radio::ble::Radio`](https://docs.rs/embassy-nrf/latest/embassy_nrf/radio/ble/struct.Radio.html)'s API surface.

The documentation for [`Radio::receive`](https://docs.rs/embassy-nrf/latest/embassy_nrf/radio/ble/struct.Radio.html#method.receive) says this:
- "If the length byte in the received package is greater than the buffer length the radio will write memory out of the buffer bounds."

The documentation for [`Radio::transmit`](https://docs.rs/embassy-nrf/latest/embassy_nrf/radio/ble/struct.Radio.html#method.transmit) says this:
- "If the length byte in the package is greater than the buffer length the radio will read memory out of the buffer bounds".

That suggests that these functions can induce accesses to out-of-bounds memory, which is usually considered [undefined behavior](https://doc.rust-lang.org/reference/behavior-considered-undefined.html).

Why, then, are these functions safe?

In the interest of giving this a fair shake, initially, I tried to justify this to myself as an "outside the abstract machine" operation. But these calls are fundamentally initiated Rust code: we don't necessarily care about how the access happens if the code in control is the same Rust program. More importantly, these functions can cause accesses to memory currently held by Rust and being used for its purposes, including memory considered immutable by Rust.

That is because, being slices, these slices can be on the stack. Depending on where that slice is, the mutation of a slice that is mutable can affect other bytes on the stack. This includes bytes that have had their accesses hoisted due to being previously taken as a reference in the program. LLVM tends to put stack bytes adjacent to each other, though the exact order can vary, especially so as LLVM can also merge stack slots under a number of conditions. And that can mean that these functions can modify bytes that are later read by Rust as a *completely unrelated* stack value, even one that has not been `let`-bound in the function yet. So I don't think that my attempt to justify this as "outside the Rust AM" can work.

The only other option I can really think of that justifies making these functions safe is that the documentation is somehow simply incorrect.

If this is doing everything bad that we would not want to have happen, without any requirement for `unsafe {}` to be used, then the API is unsound. And that is *before* getting into the function being async, which can make everything much more "entertaining". If this is unsound, then we have two obvious paths to fixing these:
- make these functions `async unsafe`
- if the byte for length really is only encoding a `u8` of length (so a maximum of 256 bytes can be received or transmitted) then the slices can be replaced with a newtype that wraps them and requires an `unsafe` constructor, guaranteeing these buffers are big enough to make the function safe, at the cost of moving the unsafety to the buffer construction, which can be significantly more ergonomic in some cases and significantly less in others

There's probably other options that require more knowledge of the implementation. I haven't futzed around with my NRF dev kit much yet.

Dirbaio, 2025-04-28:
Yes, this is unsound no doubt.

The hardware does have a way to set the max len. See https://docs.nordicsemi.com/bundle/ps_nrf52840/page/radio.html#ariaid-title22

The current code just sets it to max. https://docs.nordicsemi.com/bundle/ps_nrf52840/page/radio.html#ariaid-title22

Best fix would probably be to use that properly to ensure no overflows. It's not trivial because the hardware is writing a complex structure instead of just a buffer and the total length depends on the many fields. Maybe it'd be better with a repr(C) struct instead of a `[u8]`.

Also note the `embassy_nrf::radio::ble` module is very incomplete and not usable beyond just basic advertising. If you want full BLE support check out [trouble](https://github.com/embassy-rs/trouble) (uses Nordic's newer blob that speaks HCI) and [nrf-softdevice](https://github.com/embassy-rs/nrf-softdevice) (uses Nordic's older blob. more stable but in maintenance mode)

workingjubilee, 2025-04-30:
I have also seen a comment that suggests that there is a use-after-free somewhere in the API, but I do not know where it is since they did not pinpoint it. I am noting it publicly, even though it is essentially speculative at this moment, because it is a sufficiently serious unsoundness that any effort to poke through the files and fix things should keep an eye out for it.

commit c133ffbe (2025-05-01): Remove embassy_nrf::radio::ble.

Fixes #4144
commit df84dfec (2025-05-01): Remove embassy_nrf::radio::ble.

Fixes #4144
closed 2025-05-01 by Dirbaio