embassy-rs/embassy #4145: How to use embassy with libs that depends on std Rust

gustavowd, 2025-04-29:
I'm trying to use quick-xml in a project with embassy and reqwless.
However, several erros occur due to quick-xml usage of std Rust.

For example:
error[E0412]: cannot find type `Option` in this scope
    --> /home/gustavo/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/serde-1.0.219/src/private/ser.rs:1129:31
     |
1129 |     fn serialize_seq(self, _: Option<usize>) -> Result<Self::SerializeSeq, Self::Error> {
     |                               ^^^^^^ not found in this scope
     |
help: consider importing this enum
     |
1    + use core::option::Option;
     |

error[E0412]: cannot find type `Result` in this scope
    --> /home/gustavo/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/serde-1.0.219/src/private/ser.rs:1129:49
     |
1129 |     fn serialize_seq(self, _: Option<usize>) -> Result<Self::SerializeSeq, Self::Error> {
     |                                                 ^^^^^^ not found in this scope
     |
help: consider importing one of these items
     |
1    + use core::fmt::Result;
     |
1    + use core::result::Result;
     |
1    + use alloc::fmt::Result;
     |

error[E0425]: cannot find function, tuple struct or tuple variant `Err` in this scope
    --> /home/gustavo/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/serde-1.0.219/src/private/ser.rs:1130:9
     |
1130 |         Err(Self::bad_type(Unsupported::Sequence))
     |         ^^^ not found in this scope
     |
help: consider importing this tuple variant
     |
1    + use core::result::Result::Err;
     |

error[E0412]: cannot find type `Result` in this scope
    --> /home/gustavo/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/serde-1.0.219/src/private/ser.rs:1133:43
     |
1133 |     fn serialize_tuple(self, _: usize) -> Result<Self::SerializeTuple, Self::Error> {
     |                                           ^^^^^^ not found in this scope
     |
help: consider importing one of these items
     |
1    + use core::fmt::Result;
     |
1    + use core::result::Result;
     |
1    + use alloc::fmt::Result;

Any thoughts about this?

okhsunrog, 2025-05-01:
@gustavowd you can't use std crates on no_std, it's not related to embassy. go to crates.io and look for no_std xlm crates:
https://crates.io/search?q=%23no_std%20xml

lulf, 2025-05-26:
Closing, does not seem directly related to embassy.