embassy-rs/embassy #4158: STM32H533RE timer 12, 13 and 14 missing

moritz-geier, 2025-05-02:
Hello,
I'm currently trying embassy for the first time and wanted to use the TIM12 for some PWM, but then I saw
that it wasn't available, and also in the docs it seems to be missing. As well as TIM13 and 14.

Is this by choice or are they just missing? And is it possible to implement them?

example code
```rust
#![no_std]
#![no_main]

use embassy_executor::Spawner;
use embassy_stm32::{gpio::OutputType, time::hz, timer::complementary_pwm::{ComplementaryPwm, ComplementaryPwmPin}, timer::simple_pwm::{PwmPin, SimplePwm}};

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let p = embassy_stm32::init(Default::default());
    // rust-analyze fills in following type for pb15: PwmPin<'_, TIM12, Ch1>, so it should exist?
    let pb15 = PwmPin::new_ch2(p.PB15, OutputType::PushPull);

    // p.TIM12 is not recognized
    let pwm12 = SimplePwm::new(p.TIM12, Some(pb14), Some(pb15), None, None, hz(50), embassy_stm32::timer::low_level::CountingMode::EdgeAlignedUp);
}
```
Thanks in advance:))

Dirbaio, 2025-05-02:
TIM12 is most likely picked up by the time driver. Try using another timer by enabling the right Cargo feature, such as `time-driver-tim2`.

TIM13, TIM14 don't exist on that chip.

moritz-geier, 2025-05-03:
Thanks that worked:)

closed 2025-05-03 by moritz-geier